react-native-camera/react-native-camera #452: Camera Freeze after second Invocation of Camera in Simulator

K-Leon, 2016-10-09:
I stumbled over a interesting Bug while testing the newest master.

If i launch Camera on Simulator ( ios 9 and 10 ) the second Time and i'm using child components `<Camera><View>.....</View></Camera>` the Camera Screen is freezed on the second Invocation without throwing a crash.

Anyone experienced this before?

K-Leon, 2016-10-09:
Offending Source: https://github.com/lwansbrough/react-native-camera/blob/master/ios/RCTCameraManager.m#L38

jqn, 2016-11-14:
I ended up using this component instead https://github.com/ideacreation/react-native-barcodescanner/issues/27. It's only good for those who only need a barcode scanner. It seems the problem was resolved.

ragamufin, 2016-11-27:
I had this problem as well, I would not be able to take a picture on the simulator after the 2nd time. By removing any child elements of `Camera` it started working. Thanks! Now I put the camera in its own view with absolute positioning to fill the screen and in a view below it add the things like the take picture button that I need, with a transparent background

mankins, 2016-12-19:
See https://github.com/lwansbrough/react-native-camera/pull/523 this fixes the issue for me. Hopefully it will get merged soon.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco