react-native-camera/react-native-camera #456: Memory Leak iOS

chadwilken, 2016-10-14:
### Steps to reproduce
1. Have `Camera` component on a view
2. Open the view using a push
3. Navigate back to parent view
4. Re-navigate to the view with the Camera page and if you have a scrollable view inside that camera view with low opacity and scroll it you can see the same view behind it from the leak.

### Expected behaviour
It should destroy the camera upon component unmounting

### Actual behaviour
It retains the camera and the entire view.

### Environment
- Node v5.11.1
- RN 0.32.1
- react-native-router-flux
- iOS 9.2

### react-native-camera
**Version**: "master"

Okipa, 2016-10-27:
+1 on iOs 10.0 too

HofmannZ, 2016-12-08:
+1
iOS: 10.1.1 
React: 15.4.1
React-native: 0.39.0
React-native-camera: master on git

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco