react-native-camera/react-native-camera #457: A way to understand if the camera is loaded?

Zweer, 2016-10-15:
Hi,

is there a way to understand when the camera loads??
I've an app where you tap "record a video" in a view, then it switches to the camera -> 3-2-1-RECORD -> 10 seconds of record -> "thank you" view.

My problem is I don't know where to trigger the "3-2-1". Right now I wait 4 seconds, but it's not perfect: sometimes is 6 seconds, sometimes is 2 seconds...

Is there an event / callback / method to call??

Thanks a lot!!!
Zwe

GabiNir, 2016-10-25:
+1
I need this as well for exactly the same feature.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco