react-native-camera/react-native-camera #459: Parse.File Upload - Parse.com

MM-Psiiirus, 2016-10-20:
Hiya,
I was wondering if there is a best practice way to use the Parse.File upload of the Parse.com-SDK.
It seems a bit odd to me to convert the image into base64 and send it to the server.

https://parseplatform.github.io/Parse-SDK-JS/api/classes/Parse.File.html

Is anyone using the Parse.com Server + react-native-camera ?

killerwolf, 2017-02-13:
interested here too

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco