react-native-camera/react-native-camera #460: Camera is still active when user locks the device

tadas-subonis, 2016-10-20:
### Steps to reproduce
1. Go to a scene with CameraView so you can see the picture Preview
2. Press turn off button on the device
3. Check logs with adb logcat and you will see that camera is still active (at least on LG G3)

### Expected behaviour
Camera preview shouldn't be active

### Actual behaviour
Camera preview is active while phone is "turned off" and keeps the battery draining

Actually, when does the "preview mode" get turned off? When the component is unmounted?

### Environment
- **Node.js version**:
- **React Native version**:react-native 0.35
- **React Native platform + platform version**: react-native 0.35, Android 5.0

### react-native-camera
**Version**: master

proProbe, 2016-11-04:
We are currently killing the camera when we go to another scene. This is not optimal as the camera needs  to mount again when we enter the camera scene. This leads to serious speed drops on android (is kinda OK on ios though). Best would be if its somehow possible to paus or just turn off the camera without destroying the whole view.

baskan, 2016-11-09:
I am wondering the same question:  Actually, when does the "preview mode" get turned off? When the component is unmounted?

When our app shows the camera once we cannot make it stop capturing.

kesha-antonov, 2016-11-12:
@proProbe Hey man! How do you "kill" camera?

proProbe, 2016-11-12:
@kesha-antonov we just destroy the view. basically we just write
```
if (notCameraView) {
    return null;
} else {
    return this._renderCamera();
}
```

kesha-antonov, 2016-11-12:
Yeah, thanks! Tried that.

But I have subview:

```
render() {
  <View>
    {
      if shoulRenderCamera
        <Camera ... />
      else
        <Something else ... />
    }

  </View>
}
```

And this is not working.

carleryd, 2017-01-26:
This is still a big deal for Android devices because re-creating the Camera component whenever switching to and from the camera view in my app is very slow. This same logic run super-fast on an iPhone device however...
I am using react-native-router-flux and have a tab dedicated to the camera component and use this logic
```
      const shouldRenderCamera = this.props.redux_permission_status.camera === "authorized"
          && this.props.redux_router_scene.name === "createView"
          && (this.state.currentAppState !== "background");

      return shouldRenderCamera === true
          ? <Camera
              ref={this._setCamera}
              style={styles.preview}
              aspect={this.state.cameraConfig.aspect}
              captureTarget={Camera.constants.CaptureTarget.disk}
              captureAudio={true}
              type={this.state.cameraConfig.type}
              flashMode={this.state.cameraConfig.flashMode}
              mirrorImage={(this.state.cameraConfig.type === Camera.constants.Type.front)}
              captureQuality={Camera.constants.CaptureQuality.high}
              defaultOnFocusComponent={true}
              // Makes tap to focus available
              onFocusChanged={this._focusChanged}
              // Makes zoom available
              onZoomChanged={this._zoomChanged}
          />
          : null
```

gregfedorov, 2017-07-07:
I've had exact same issue and approached it the AppState way but by extending the Camera component.

class BetterCamera extends Camera {
	constructor(props) {
		super(props)
		this.state = {
			appState: AppState.currentState
		}
	}

	componentDidMount() {
		AppState.addEventListener('change', this._handleAppStateChange)
		if (typeof super.componentDidMount === "function") { 
			super.componentDidMount()
		}
	}

	componentWillUnmount() {
		super.componentWillUnmount()
		AppState.removeEventListener('change', this._handleAppStateChange)
	}

	_handleAppStateChange = (nextAppState) => {
		this.setState({appState: nextAppState});
	}

	render() {
		if(this.state.appState === 'active') {
			return(
				super.render()
			)
		} else {
			return null
		}
	}
}

djalmaaraujo, 2017-12-27:
@gregfedorov That worked like a charm.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco