react-native-camera/react-native-camera #461: Deadlock when rotating app

johnryan, 2016-10-22:
### Steps to reproduce
Create a camera view and rotate the device several times.

### Expected behaviour
The preview should rotate with the app.

### Actual behaviour
Eventually the preview layer gets locked in RCTCamera.m -layoutSubviews and the main thread becomes locked. I've attached a screenshot of the stack trace. I was able to solve the problem by changing layoutSubviews but it seems like a hack:

```
- (void)layoutSubviews
{
  [super layoutSubviews];
    dispatch_async(dispatch_get_main_queue(), ^{
        self.manager.previewLayer.frame = self.bounds;
        [self setBackgroundColor:[UIColor blackColor]];
        [self.layer insertSublayer:self.manager.previewLayer atIndex:0];
    });
  
}
```

[image]

### Environment
- **Node.js version**: 6.8.0
- **React Native version**: 0.35
- **React Native platform + platform version**: iOS 9.3.5

### react-native-camera
**Version**: master

Anuj-786, 2016-10-26:
I think you should have to use orientation i.e 
Values: `Camera.constants.Orientation.auto or "auto" (default), Camera.constants.Orientation.landscapeLeft or "landscapeLeft", Camera.constants.Orientation.landscapeRight or "landscapeRight", Camera.constants.Orientation.portrait or "portrait", Camera.constants.Orientation.portraitUpsideDown or "portraitUpsideDown"`

johnryan, 2016-10-26:
I haven't changed any of the values so it is set on the default (auto)  and as such should rotate with the rest of the app. It works sometimes but eventually locks up and freezes the app.

GeoffreyPlitt, 2016-10-31:
I have a stacktrace which may help:

java.lang.RuntimeException·getParameters failed (empty parameters)
Raw
Camera.java:-2android.hardware.Camera.native_getParameters	
Camera.java:1996android.hardware.Camera.getParameters	
RCTCamera.java:332com.lwansbrough.RCTCamera.RCTCamera.adjustPreviewLayout	
RCTCamera.java:169com.lwansbrough.RCTCamera.RCTCamera.setActualDeviceOrientation	
RCTCameraView.java:122com.lwansbrough.RCTCamera.RCTCameraView.setActualDeviceOrientation	
RCTCameraView.java:16com.lwansbrough.RCTCamera.RCTCameraView.access$100	
RCTCameraView.java:34com.lwansbrough.RCTCamera.RCTCameraView$1.onOrientationChanged	
OrientationEventListener.java:403android.view.OrientationEventListener$SensorEventListenerImpl.onSensorChanged	
SystemSensorManager.java:583android.hardware.SystemSensorManager$SensorEventQueue.dispatchSensorEvent	
MessageQueue.java:-2android.os.MessageQueue.nativePollOnce	
MessageQueue.java:323android.os.MessageQueue.next	
Looper.java:143android.os.Looper.loop	
ActivityThread.java:7229android.app.ActivityThread.main	
Method.java:-2java.lang.reflect.Method.invoke	
ZygoteInit.java:1230com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run	
ZygoteInit.java:1120com.android.internal.os.ZygoteInit.main

GeoffreyPlitt, 2016-10-31:
@johnryan Should I open a 2nd issue? I didn't see "ios" in the issue title. It's the exact same thing on Android - a deadlock when orientation flips.

johnryan, 2016-10-31:
Ah ok, could be related then, I assumed it was a native iOS issue

skizzo, 2017-08-02:
+1, was anybody able to solve this? this bug makes the component pretty much unusable.. thanks

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

neogermi, 2018-02-01:
Hello, this is still a common issue for us and reproducible. The fix of the forks above seems to help though. Is there a PR for this?

neogermi, 2018-02-01:
If there is no PR, I can create one if you want.

sibelius, 2018-02-01:
Send a pr please

neogermi, 2018-02-01:
See https://github.com/react-native-community/react-native-camera/pull/1194 :)

enieber, 2018-08-30:
@sibelius or @jgfidelis, this issue can it is being closed?

closed 2018-11-20 by n1ru4l