react-native-camera/react-native-camera #466: Crashes under iOS10 simulator

freshlybakedcode, 2016-10-30:
### Steps to reproduce
1. Start a new react native project
2. Install react-native-camera following instructions
3. Add example source code (BadInstragramCloneApp)
4. Change export statement from 'AppRegistry.registerComponent...' to 'module.exports = (BadInstagramCloneApp);'
4. Run react-native ios-start from CLI
5. Launches using IOS10 simulator

### Expected behaviour
Camera function launched; '[CAPTURE]' button displayed on black screen

### Actual behaviour
Crashes out of simulator, nothing logged to console. 'Pause on all caught exceptions' shows:
**_nodeUtil.js**
/** Used to access faster Node.js helpers. */
var nodeUtil = (function() {
  try {
    return freeProcess && freeProcess.binding('util');
  } catch (e) {}
}());

If you choose to run the same code in the IOS9 simulator then it works as expected.

### Environment
- **Node.js version**: 6.8.1
- **React Native version**:  CLI: 1.0.0, React native: 0.36.0
- **React Native platform + platform version**: iOS 10.0, iPhone 6, 50% scaling

### react-native-camera
**Version**: react-native-camera@https://github.com/lwansbrough/react-native-camera.git

JustinNothling, 2016-10-31:
Getting the same issue

derekalia, 2016-11-01:
With iOS 10 and higher you need to add the "Privacy - Camera Usage Description" key to the info.plist of your project.

pvpshoot, 2016-11-01:
Yes, i just read, but app still crushed

freshlybakedcode, 2016-11-01:
Adding the following still results in iOS 10 simulator falling over:

```
<key>NSPhotoLibraryUsageDescription</key>
<string>Any string you want</string>
```

(As noted in step 4 of the original issue report)

boblad, 2016-11-01:
This link helped me fix my issue. http://stackoverflow.com/questions/38236723/ios-10-error-access-private-when-using-uiimagepickercontroller

freshlybakedcode, 2016-11-01:
Okay! So looks like this is the key you actually need:

`<key>NSCameraUsageDescription</key>
<string>$(PRODUCT_NAME) camera use</string>`

I have that in conjunction with the NSPhotoLibraryUsageDescription and iOS10 simulator works - but only when it's run from within Xcode for some reason.  The usual `react-native run-ios` from the command line still crashes, but building and running from Xcode isn't the end of the world.

closed 2016-11-03 by freshlybakedcode

mersay, 2016-11-16:
this is also happening on real device, iOS 10.1.1 iPad mini 4

freshlybakedcode, 2016-11-16:
Interesting; my app's now live on the App store and is working fine on an iPhone 6 with iOS 10.1.1; I don't have an iPad to try it with though.  Do you know if yours crashes on an iPhone?

ywongweb, 2016-11-18:
Adding both `NSCameraUsageDescription` and `NSPhotoLibraryUsageDescription` fixed the problem for me, thanks @freshlybakedcode!
Tested on iPad iOS 10.1.1

freshlybakedcode, 2016-11-18:
@ywongweb NICE! Pleased to hear that you got it going - I'm going to get a pull request sorted to update the docs for other users.

stereodenis, 2018-01-07:
In my case it is `NSPhotoLibraryAddUsageDescription`, not `NSPhotoLibraryUsageDescription`

ospfranco, 2018-01-07:
you need to add the corresponding strings to your plist file

Oscar Franco
Software Engineer @ Mantro GmbH
Munich, Germany
+4915123657301
oscar.franco@mantro.net | ospfranco@gmail.com