react-native-camera/react-native-camera #471: Freeze on Andorid

baskan, 2016-11-07:
### Steps to reproduce
1. npm install react-native-camera@https://github.com/lwansbrough/react-native-camera.git --save
2. place Camera element.

### Expected behaviour
Should at least log something

### Actual behaviour
App freezes when I try to call capture() (runs fine in iOS)
After the freeze camera hangs and cannot reach camera even in native camera app.

### Environment
- **Node.js version**: v6.3.1
- **React Native version**: 0.34.1
- **React Native platform + platform version**: Android 6.0

### react-native-camera
**Version**: "master"

baskan, 2016-11-07:
Aight I figure this out now. If u experience unexpected freezes on android downgrade your **build tools version 24.0.3 works fine** 25.0.0 is not.

I still get 

> E/WVMExtractor: Failed to open libwvm.so: dlopen failed: library "libwvm.so" not found

but works.

Im closing this with hoping to help someone some day :)

closed 2016-11-07 by baskan

ragamufin, 2016-11-27:
Thanks, this solved my issue!