react-native-camera/react-native-camera #472: Can't link to camera to app

IvanProdaiko94, 2016-11-12:
I have done all the steps that are required to link camera, but when I start my app it crashes on the building phase with this error (permission are also [added](https://monosnap.com/file/KTa5vVa2T6pPYCmCRBERLjWJfq3nsa)

leechance, 2016-11-14:
ios or android? my android demo is crashed

IvanProdaiko94, 2016-11-14:
Ios(

IvanProdaiko94, 2016-11-14:
react-native 0.36.1

ngsiolei, 2016-11-16:
@IvanProdaiko94 the screen capture in your first post seems to be something about Camera Roll access, the property `captureTarget` is default to `Camera.constants.CaptureTarget.cameraRoll` in ios (more details in README), see if this is your case.

IvanProdaiko94, 2016-11-28:
So you think that the problem in permissions?
I added one, as it was mentioned in docs, but it fails on build img phase понедельник,

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco