react-native-camera/react-native-camera #473: Is this maintaned?

kesha-antonov, 2016-11-12:
Hello guys!

Thanks for the great module! Love it.

Please tell me is this currently (actively) maintained? I see issues are raising and lots of PRs are not accepted.
Maybe author does not have enough time and we can somehow help?

btoueg, 2016-11-15:
I'd like to know too, because there are so many improvement waiting to be merged, it can't stay inactive like this.

charpeni, 2016-11-16:
Hey there!

Sure this is maintained ! Unfortunately, we're a small number of collaborators with limited hours.

Issues are growing up with feature requests, bugs and questions. On the other side, PRs are growing up with new features and attempt to fix issues without test plan or a minimal one.

It's hard for us to actively review all pull requests, we don't want to merge new features without a solid test plan where we're sure of the stability on all devices. Same thing happened with bug fix, should we merge it without a proper test from the community ? Sure it can fix an issue, but it can cause more.

Help and PRs are welcome. :)

If you find a pull request that can and should be merge, please ping it would be a pleasure.

rt2zz, 2016-11-17:
Indeed, please note we badly want to get camera to v1! There is however a lot of complexity under the hood and we need help making sure each change is good for the long term and not just a short term solution.

closed 2016-11-17 by rt2zz

kesha-antonov, 2016-11-17:
Hey!

Got it. Thanks for your work, really appreciated!