react-native-camera/react-native-camera #475: android image rotation is wrong (mirrorImage=true, orientation=portait)

madsleejensen, 2016-11-14:
### Steps to reproduce
1. Take a photo with Front camera, with `mirrorImage=true` `orientation=portait`

### Expected behaviour
The image should be captured as the preview view displays.

### Actual behaviour
The image is rotated 90deg. The expected behavior would be the same rotation as `mirrorImage=false` but flipped horizontal.

### Environment
- react-native v0.34.0
- **React Native platform + platform version**: Android 6.x (Samsung) 

### react-native-camera
**Version**: master

jarradh, 2016-11-24:
We're experiencing the same behaviour

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco