react-native-camera/react-native-camera #476: We receive multiple (~10x) same barcode in a single camera capture.

ursenix, 2016-11-15:
### Steps to reproduce
1. Show camera on the barcode
2. You receive data multiple times for a single attempt.

### Expected behaviour
We should able to stop the capturing, we need an event to stop and start the barcode capture.

### Actual behaviour
We receive multiple (~10x) same barcode in a single camera capture. We are unable to stopCapturing the barcode.

### Environment
- **Node.js version**: 6.7
- **React Native version**: 0.36
- **React Native platform + platform version**: iOS 10.0

### react-native-camera
**Version**: 0.3.8

ngsiolei, 2016-11-16:
How about to have a flag variable for barcode handling? For example, if a desired barcode is detected, set the flag and ignore the rest barcode capture. Later on, reset the flag to resume.

jqn, 2016-11-16:
@ursenix - that is the default functionality of the scanner and there is no stop capture method. Like @ngsiolei said setting a flag would take care of this. Or checkout this issue https://github.com/ideacreation/react-native-barcodescanner/issues/46 and you could try what @someok commented on.

ursenix, 2016-11-17:
Thanks for the response.
We succeeded with flag setting and setTimeout to clear the flag.
But the issue is when we navigate to other pages and come back, the setState changes is not rendered in view.

ngsiolei, 2016-11-17:
In my project, I don't want to rerender the view upon flag (the boolean variable for handling barcode or not) change, so flag is saved as class variable, instead of as state. I initialize the flag in `constructor()`, update (set/reset) the flag in `onBarCodeRead` handler. I always remind myself to reset the flag to its initial value before leaving current view, so the barcode handling still works normally if the view cannot be unmounted for some reasons (I must admit that I don't understand app navigation well).

prabinyovan, 2016-11-23:
We had the similar problem and we fixed that with the following trick.

1. Unbind event listener ie Call componentWillUnmount() method immediately after the data read from barcode.
2. Rebind event listener ie Call componentWillMount() method after some delay. 

Below is my modified _onBarCodeRead function

```
_onBarCodeRead = (data) =>{    
                 if (this.props.onBarCodeRead) {
                        this.componentWillUnmount()
                       this.props.onBarCodeRead(data)
                       setTimeout(() => {
                                 this.componentWillMount();
                      }, 3000);
               }
        };
```

ursenix, 2016-11-23:
Thanks @prabinyovan. It's working as charm. You saved my day!

closed 2016-11-23 by ursenix

KGALLET, 2017-03-25:
Thx @prabinyovan

mikeboy91, 2017-05-04:
I am facing the same problem, but the trick from @prabinyovan doesn't work for me.
Did I something wrong? I'm not that experienced with react-native yet.

```
class QRScanner extends Component {

  constructor(props) {
    super(props);

    this.camera = null;

    this.state = {
      camera: {
        aspect: Camera.constants.Aspect.fill,
        captureTarget: Camera.constants.CaptureTarget.cameraRoll,
        type: Camera.constants.Type.back,
        orientation: Camera.constants.Orientation.auto,
        flashMode: Camera.constants.FlashMode.auto,
        onBarCodeRead: this._onBarCodeRead
      }
    };
  }

  componentWillMount() {
    this.setState({onBarCodeRead: this._onBarCodeRead});
  }

  componentWillUnmount() {
    this.setState({onBarCodeRead: null});
  }

  _onBarCodeRead = (event) => {
    this.componentWillUnmount();
    this.props.handleBarCode(event);
    setTimeout(() => {
      this.componentWillMount();
    }, 3000);
  }

  render() {
    return (
      <View style={styles.container}>
        <Camera
          ref={(cam) => {
            this.camera = cam;
          }}
          style={styles.preview}
          aspect={this.state.camera.aspect}
          captureTarget={this.state.camera.captureTarget}
          type={this.state.camera.type}
          flashMode={this.state.camera.flashMode}
          onBarCodeRead={this.state.camera.onBarCodeRead}
        >
        </Camera>   
      </View>
    );
  }
}
```

fbacker, 2017-09-25:
~~~~
componentDidMount() {
      this._codes = new Array();
}

barcodeReceived(e) {
      console.log('Barcode: ' + e.data);
      console.log('Type: ' + e.type);
      if(e!=null && (e.type==Camera.constants.BarCodeType.datamatrix||e.type==Camera.constants.BarCodeType.qr)){
        let code = e.data;
        if(!this._codes.includes(code)){

          if(Platform.OS === 'ios'){
            Vibration.vibrate(500, false);
          }
          else{
            Vibration.vibrate([0, 500], false);
          }

          this._codes.push(code);
          this.props.onCode(code);
        }
      }
    }
~~~~

miguelacio, 2018-05-02:
I was able to "fix" this issue by doing the following: 
``` class CustomCamera extends Component {
  state = {
    aspectRatio: '4:3',
    changeScreen: false,
  }
  onhandleQRCode = qrData => {
    const { changeScreen } = this.state
    try {
      if (!changeScreen) {
        this.setState({ changeScreen: true })
        const { navigation: { navigate } } = this.props
        const { data } = qrData
        const parsedData = JSON.parse(data)
        const ids = parsedData.map(({ id }) => id)
        navigate('CashCollection', { ids })
      }
    } catch (error) {
      alert(error)
    }
  }
     render() {
    const { aspectRatio } = this.state
    return (
      ...
  }