react-native-camera/react-native-camera #484: When set aspect to fit/fill, The preview image has incorrect ratio

peter4k, 2016-11-20:
### Steps to reproduce
1. Create a camera preview with aspect fit

### Expected behaviour
when set to fit, the preview image will has a width:height 16:9. But the image is still 4:3, which makes the image looks stretched.

### Actual behaviour
The preview image is stretched, looks like the library thinks the ratio is 16:9 but it is actually 4:3

### Environment
- **Node.js version**:7.1.0
- **React Native version**: 0.35
- **React Native platform + platform version**: Android 5.0

### react-native-camera
**Version**: master

Fit Aspect:
[image]
Stretch Aspect (the preview component has 4:3 ratio):
[image]

sibelius, 2017-12-12:
is this still happening?

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco