react-native-camera/react-native-camera #485: It doesn't match with iphone camera's default ratio

mnkang89, 2016-11-23:
### Steps to reproduce
1. I just followed the default example in react-native-camera.

### Expected behaviour
has the same ratio(or size) with native camera app (when I ran on iphone6)

### Actual behaviour
It doesn't match with default iphone camera app's ratio. it felt like react-native-camera's view is more zommed.. so i adjusted some parameters but it's not work..
[image]
[image]

umesh-magicpin, 2016-11-23:
facing the same issue.

machadogj, 2016-11-25:
I am experiencing the same problem. Any progress on this yet?

antonshevchenko, 2016-11-25:
Noticed this too. It looks like it defaults to the "video" setting (if you got to the default camera app and swipe to "video" mode). Also, Snapchat has the same "zoom" as on the video mode. Not sure if/how this can be changed.

mnkang89, 2016-11-26:
thx guys!! i hope this issue will be fixed soon lol!

ionflow, 2016-11-27:
I'm experiencing the same issue on iPhone 6 as well. 

Also noticed the picture quality does not match the native camera regardless of what quality I choose in the settings. And the camera exposure does not adjust to match the native camera when shooting the same exact target/scene.

almost, 2017-04-26:
Did anyone find a workaround for this issue?

VikramK, 2017-12-05:
This setting worked for me. 

this.state = {
      camera: {
                     captureQuality: Camera.constants.CaptureQuality.photo
       }

  <Camera
                ref="cam"
                style={styles.container}
                type={this.state.cameraType}
                captureQuality={this.state.camera.captureQuality} />

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco