react-native-camera/react-native-camera #489: Can't display camera feed on ios

peterept, 2016-11-28:
I can't get react-native-camera to work in my app, nor can I get the Example project.

## If I try the Example project:
1. Download the source
2. $ cd Example
3. $ npm init
4. $ react-native run-ios
### Expected: 
app runs with camera feed
### Actual: 
I get red error screen: Unhandled JS Exception: Invariant Violation: Module name prefixes should've been stripped by the native side but wasn't for RCTJSCExecutor

## my IOS project
1. Follow the instructions for IOS:
2. $ npm install react-native-camera@https://github.com/lwansbrough/react-native-camera.git --save
3. $ react-native link react-native-camera
4. Setup xcode library references
5. Update code in index.ios.js to require and see camera view
### Expected: 
camera feed
### Actual: 
I get red error screen: undefined is not an object (evaluating 'CameraManager.Aspect')

I've tried both master branch and 0.4.0 branch.

Versions:
- react-native-cli: 1.2.0
- react-native: 0.34.1
- npm version: 3.10.9

Any help or pointers greatly appreciated !!

Peter

sashakid, 2017-01-31:
See already opened issue https://github.com/lwansbrough/react-native-camera/issues/528

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco