react-native-camera/react-native-camera #490: When I take pictures too fast, the app crashed

Checccy, 2016-11-28:
### Steps to reproduce
1.When I take pictures too fast,
2.When I change the camera type too too fast

### Expected behaviour
still running & take photo &change type

### Actual behaviour
the app crashed,
it show my project stop running

### Environment
- **Node.js version**:latest
- **React Native version**:latest
- **React Native platform:Android

### react-native-camera
**Version**: >0.40

duzliang, 2016-12-14:
I'm also have the same issue
react-native: 0.38
react: 15.4.1

stack trace:

### java.lang.RuntimeException·takePicture failed

`Camera.java:-2android.hardware.Camera.native_takePicture	
Camera.java:1531android.hardware.Camera.takePicture	
Camera.java:1470android.hardware.Camera.takePicture	
RCTCameraModule.java:508com.lwansbrough.RCTCamera.RCTCameraModule.captureWithOrientation	
RCTCameraModule.java:50com.lwansbrough.RCTCamera.RCTCameraModule.access$100	
RCTCameraModule.java:474com.lwansbrough.RCTCamera.RCTCameraModule$2.orientationEvent	
RCTSensorOrientationChecker.java:64com.lwansbrough.RCTCamera.RCTSensorOrientationChecker$Listener.onSensorChanged	
SystemSensorManager.java:427android.hardware.SystemSensorManager$SensorEventQueue.dispatchSensorEvent	
MessageQueue.java:-2android.os.MessageQueue.nativePollOnce	
MessageQueue.java:143android.os.MessageQueue.next	
Looper.java:122android.os.Looper.loop	
ActivityThread.java:5298android.app.ActivityThread.main	
Method.java:-2java.lang.reflect.Method.invoke	
Method.java:372java.lang.reflect.Method.invoke	
ZygoteInit.java:913com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run	
ZygoteInit.java:708com.android.internal.os.ZygoteInit.main`

deviousgeek, 2016-12-22:
Same issue here, perhaps the best we can do here is rate limit the capture button

CreatineDevelopment, 2017-01-11:
I have addressed the capture issue in pull request #553

I was not aware of the camera switch issue at the time, so it is not fixed in this pull request.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco