react-native-camera/react-native-camera #492: Detect QRcode/Barcode on Android is really slow

pmualaba, 2016-11-29:
### Steps to reproduce
1. Try to Scan QRcodes/barcodes on Android
2. When the phone moves a little it has difficulty to detect  QRcodes/barcodes
3. You need to hold the device stable and unmoved for the QRcodes to be detected (sometimes more then 5 seconds)
4. This behaviour is only on Android, iOS works perfect.

### Expected behaviour
Quickly detect QRcodes/barcodes on Android, even when the phone moves a little bit. On iOS QRcodes/barcodes are detected immediately even when the phone is moving.

### Actual behaviour
You have to wait more then 5 sec. and firmly hold your phone stable to be able to scan QRcodes/barcodes on Android

### Environment
- **Node.js version**: 6+
- **React Native version**: 0.38
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc 
Android 5.1.1

### react-native-camera
**Version**: npm version or "master"
master

tsanthosh, 2017-01-02:
Having the same issue with Android. For me its about 2 -3 seconds. Works great on iOS.

dinhha, 2017-06-20:
Having them same issue. Are there any improvement?

augusto-altman, 2017-11-04:
Same issue here, is there any news regarding this?

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

ospfranco, 2018-01-12:
Just as a bonus for all of you guys, try the new implementation on the master branch it is called RNCamera, let us know if it works it is based on the expo framework library

closed 2018-01-12 by ospfranco