react-native-camera/react-native-camera #494: Zoom and Focus (ios and android)

scerelli, 2016-11-29:
Can someone please tell me if pinch to zoom and tap to focus exist in this component? It's should, based on readme, but on ios and android both functions not working.

btoueg, 2016-12-01:
pinch to zoom and tap to focus works well on Android
pinch to zoom on Android is supported on PR #498
I'm not sure for tap to focus on Android, I'd say no, my PO would say yes

scerelli, 2016-12-01:
@btoueg how can i enable it on android? 

Note: i just saw that you can enable pinch to zoom and tap to focus on IOS by passing empty function at `onFocusChanged` and `zoomChanged` callback props. I don't know why i think is a bug

scerelli, 2016-12-01:
and even if you log something inside these callbacks (onFocusChanged and zoomChanged) they never being called.

Kottidev, 2017-01-03:
+1  

onZoomChanged doesn't work on Android

ju5td0m7m1nd, 2017-01-08:
Check this PR
https://github.com/lwansbrough/react-native-camera/pull/550

I've activated the callback function but only for iOS
Need others help me out for Android

kevinnguy, 2017-01-27:
@scerelli have you submitted a pull request for your Android focus changes? It'd be super helpful!

scerelli, 2017-01-29:
Sorry @kevinnguy  I didn't open a pull request because it wasn't working really well and I don't feel really confidente with java so I don't want to merge my shitty code on this codebase that has already it's own problems.

abrahambotros, 2017-01-30:
Hi all, please check out https://github.com/lwansbrough/react-native-camera/pull/575 for tap-to-focus and improved (continuous) auto-focus. I've also noted in the README that indeed `onFocusChanged` and `onZoomChanged` don't actually get called on Android yet, though pinch-to-zoom seems to work (without any of my PR changes) and https://github.com/lwansbrough/react-native-camera/pull/575 should enable tap-to-focus and better auto-focus as you guys have been mentioning.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco

fabriziobertoglio1987, 2019-10-15:
@ospfranco 

Maybe not using react-native-camera and instead [just writing an intent to open the default camera application](https://stackoverflow.com/a/2729382/7295772)

[The following app](https://synced.to/) uses this solution, all the camera functionalities work perfectly. 

[image]