react-native-camera/react-native-camera #495: Image produced after capture has a wider viewport than that visible in the preview

go4soumya, 2016-11-29:
### Steps to reproduce
1. View a frame of specific width from the preview (as in the example provied)
2. Capture the frame
3. Objects outside the specified frame comes into the image produced

### Expected behaviour
The image should have the exact frame as that visible in the preview. The viewport should not get wider.

### Actual behaviour
The image produced after capture shows object outside the intended/captured frame. The image is usually wider than expected.

### Environment
- **React Native version**: 0.38.0
- **React Native platform + platform version**: Windows + Android 5.0

J-Priebe, 2016-12-15:
Same problem here.

jcmandersson, 2017-02-23:
Opposite problem here, I see objects in the preview that is not visible in the actual photo. Any progress on this?

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco