react-native-camera/react-native-camera #500: IOS barcode scanner show black screen when second time access.

HJ29, 2016-12-02:
### Steps to reproduce
1. navigate into scanner page
2. scan and navigate to another page
3. back to scanner page
4. black screen show instead of scanner

### Expected behaviour
Camera should open as usual

### Actual behaviour
Camera appear as a black screen

### Environment
- **Node.js version**: 6.7.0
- **React Native version**: 0.35.0
- **React Native platform + platform version**: iOS 10.0

### react-native-camera
**Version**: 0.3.8

closed 2016-12-09 by HJ29

gavaliswapnil, 2016-12-26:
Also i have issue in qr scanning.
Please anybody give me solution

Ethandf90, 2018-02-21:
same here.
To reproduce, show scan page -> QR code read -> move to next page -> .... -> scan page is black when opened again -> back out of nav stack -> enter again then it's fine