react-native-camera/react-native-camera #501: Crop images based on camera view

proProbe, 2016-12-03:
Hello!

Great library! Im stumbling upon a problem regarding cropping the taken images based on the aspect ratio that I've given to the camera. Lets say that Im rendering the camera in a view that has `height: 300, width: 400`, Im still getting images that are in full size. Is it possible to crop the images based on the camera size or is not supported and should be done with some post-process?

J-Priebe, 2016-12-12:
I too spent a lot of time trying to figure this out... after looking at a ton of cropping libraries I discovered you can do it programatically with [ImageEditor](https://facebook.github.io/react-native/docs/imageeditor.html). There is an example by Facebook [here](https://github.com/facebook/react-native/blob/1d819e9503e69737733c3a0890ae272597527083/Examples/UIExplorer/ImageEditingExample.js)

Oddly hard to find a sparsely documented, but very convenient.

blhack, 2017-05-24:
Would love to see this as a feature as well.  It seems like the most obvious behavior.

npomfret, 2017-05-24:
Be warned, the RN ImageEditor does everything in base64, which means it uses more memory than a native implementation would, which on android phones in particular might mean the app crashing.

joonhocho, 2017-08-04:
I need to do this too. Is there any image processing module for react native that can do simple crop, preferably on native side as @npomfret  mentioned? I've tried `ImageEditor` module, but it does not work with disk / temp file urls. @J-Priebe Did you get `ImageEditor` working for you with `react-native-camera`?

victorbadila, 2017-09-28:
+1 I made some workarounds with ImageEditor but it is highly unstable on Android especially. This feature should be built in, it doesn't even make sense to show a specific camera view and when you capture the picture it looks completely different, this is an actual bug and should have high priority.

mirrorshades, 2017-10-27:
So this PR was supposed to handle this but was never merged due to conflicts apparently. #627

sunweiyang, 2018-01-10:
Looks like this is addresed in #1057.

sibelius, 2018-01-11:
Done

closed 2018-01-11 by sibelius

Traviskn, 2018-02-28:
It seems in the new release we have lost the ability to use `cropToPreview`, are there any plans to get cropping to preview working on the the new RNCamera?

dipeshkoirala21, 2021-03-01:
can somebody help me out here? I need to use cropToPreview props, just like @Traviskn said. Is there any work around ?

azazel404, 2021-03-03:
this case already close , and how to solve this issue , i have some problem Crop images based on camera view , in the documentation react native camera , doesnt support cropping image right ?

azazel404, 2021-03-03:
> can somebody help me out here? I need to use cropToPreview props, just like @Traviskn said. Is there any work around ?

me too , your comment is few days ago , how about now , u already solve with this problem ?