react-native-camera/react-native-camera #502: No camera shown on Android

timbielawski, 2016-12-05:
### Steps to reproduce
1. Install via instructions
2. Add permission to manifest

   <uses-permission android:name="android.permission.CAMERA" />
    <uses-permission android:name="android.permission.RECORD_AUDIO"/>
    <uses-permission android:name="android.permission.RECORD_VIDEO"/>
    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />

### Expected behaviour
See camera on camera screen

### Actual behaviour
On Android there is no camera, IOS is fine

### Environment
-RN 0.35

timbielawski, 2016-12-06:
It says no camera found ?

timbielawski, 2016-12-07:
Ah this solves it 

https://facebook.github.io/react-native/docs/permissionsandroid.html

closed 2016-12-07 by timbielawski

wardnath, 2017-01-01:
@timbielawski could you elaborate on how you integrated `PermissionsAndroid` with the camera?

JonghunYu, 2017-02-07:
``` javascript
  const PERMISSION = PermissionsAndroid.PERMISSIONS.CAMERA;

  async function requestCameraPermission() {
    var permission = await PermissionsAndroid.checkPermission(PERMISSION);
    if (permission) return true;
    return await PermissionsAndroid.requestPermission(PERMISSION);
  }
```
FYI @wardnath 
please note that this is async function which returns promise