react-native-camera/react-native-camera #503: Translation warnings

stoffern, 2016-12-07:
### Steps to reproduce
Build for android

### Expected behaviour
no warnings

### Actual behaviour
```
warning: string 'catalyst_debugjs' has no default translation.
warning: string 'catalyst_element_inspector' has no default translation.
warning: string 'catalyst_jsload_error' has no default translation.
warning: string 'catalyst_jsload_message' has no default translation.
warning: string 'catalyst_jsload_title' has no default translation.
warning: string 'catalyst_reloadjs' has no default translation.
warning: string 'catalyst_settings' has no default translation.
warning: string 'catalyst_settings_title' has no default translation.
```

ZeroCho, 2017-08-13:
@stoffern I still have this one. Did you solved this? It's just warnings but annoying though.

afilp, 2017-10-25:
I have the same issue. Any ideas?

stoffern, 2017-10-25:
oh, dont remember why i closed it. lets reopen it then 😉

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco

FerTeran96, 2019-04-23:
I have de same problem