react-native-camera/react-native-camera #509: Receiving Photo with 2-3 seconds delay

ShvedDmutro, 2016-12-09:
### Steps to reproduce
1. make a photo
2. received photo with big delay

### Expected behaviour
Should receive photo without long delays

### Actual behaviour
When make photo, received photo with big delay.
This problem starts from commit - https://github.com/lwansbrough/react-native-camera/commit/06dc7ac5a71e3d997632ab3493e03c74defd176e

sundayhd, 2017-01-09:
100% agree

npomfret, 2017-03-21:
Hey @sundayhd  & @ShvedDmutro 

I'm made some memory and performance fixes in my fork: https://github.com/npomfret/react-native-camera/tree/performance

If you get a chance to test that would be helpful.  Will submit a PR soon.

DaBs, 2017-05-19:
@npomfret I'm having the same issues, and your performance fixes, which has now been merged to the master branch, has not helped it. Depending on the quality, I get varying speeds, low = 0,5s of delay, medium = 1,5s of delay and high = 3 seconds of delay.

I'm thinking if it's due to it being async?

npomfret, 2017-05-20:
@DaBs  3s still sounds quite bad. On what device are you seeing these delays?

sundayhd, 2017-05-20:
Something like 2.5-3 sec delay on Nexus 6p.
Thanks for your time ;-)

DaBs, 2017-05-20:
@npomfret A OnePlus 3, which is what concerns me

npomfret, 2017-05-21:
@sundayhd @DaBs how are you java skills?  are you able to put some logging in to see if you can capture where the time is going?

DaBs, 2017-05-21:
@npomfret I think I'd be capable of doing that, let me get back to you later.

DaBs, 2017-05-21:
@npomfret The logs confirms what I'm seeing - a difference hovering around 2,5 - 3 seconds after the original capture method call until it actually starts processing and then close to no time difference in processing it and saving it.
```
05-21 12:56:27.945  6862  6910 I RCTCameraModule: Activated react method capture
05-21 12:56:27.946  6862  6910 I RCTCameraModule: Started capturing image
....
05-21 12:56:29.403  6862  6862 I RCTCameraModule: Captured image, processing now
```

Is the example of the logs I've inserted, respectively at [RCTCameraModule.java#487](https://github.com/lwansbrough/react-native-camera/blob/master/android/src/main/java/com/lwansbrough/RCTCamera/RCTCameraModule.java#L487), [RCTCameraModule.java#505](https://github.com/lwansbrough/react-native-camera/blob/master/android/src/main/java/com/lwansbrough/RCTCamera/RCTCameraModule.java#L505) and [RCTCameraModule.java#584](https://github.com/lwansbrough/react-native-camera/blob/master/android/src/main/java/com/lwansbrough/RCTCamera/RCTCameraModule.java#L584). There's more logs in between that might come in handy which I'm looking at right now.

ZekkieB, 2017-05-22:
I'm having the same issue, however it works normal on a virtual device. But on my own i have a delay of a few seconds.

DavitVosk, 2017-06-20:
Same problem here. Are there any updates?

RekkyRek, 2017-07-05:
Same problem. Capturing with an emulator is almost instant as @ZekkieB said though. Currently it's really bogging down my app and is basically making it unusable :/

Tzinov15, 2017-07-10:
I'm having this exact same issue on iOS as well. Fast on simulator, upwards of 3-4 seconds sometimes on device

@ShvedDmutro you mentioned this started happening from commit - **06dc7ac** onwards. 

Maybe it's worth reverting to then if not a lot of functionality has been added since? I'm desperate for a fix on this as well

ospfranco, 2018-01-13:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-13 by ospfranco