react-native-camera/react-native-camera #510: works in debug build, not on release build ( android )

ataomega, 2016-12-09:
### Steps to reproduce
1. Works perfect on iOS debug or release. app already on app store. fine
2. works on debug release on android. but not in release build on android
3. should i change any configuration?

### Expected behaviour
camera works on debug build not on release build

### Actual behaviour
Camera doesn't work. the modal opens, i can see the rectangle and close button, but camera won't show anything on simulator or real device. but it works on debug build.

### Environment
- **React Native version**: 0.38.0
- **React Native platform + platform version**: Android 5.0

### react-native-camera
**Version**: 0.4.0

CarreraPHP, 2017-02-10:
any update on the issue? I am facing it too...

ospfranco, 2018-01-13:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-13 by ospfranco

subodhsri, 2018-12-20:
reopening.

n1ru4l, 2018-12-23:
@sbodhrsi If you experience this issue with the newest version of `RNCamera` please open a new issue with the log output.