react-native-camera/react-native-camera #512: Android Camera using Navigator No Preview(white screen)

weremax, 2016-12-12:
### Steps to reproduce
1. create a simple Navigator for android.
2. using the default code provided in the examples section modified to fit the structure.
3. set initialRoute as the step 2 above's component. All you get is a white screen on android.

### Expected behaviour
camera should appear

### Actual behaviour
White screen no camera preview.

### Environment
- **Node.js version**:
- **React Native version**: 0.39.0
- **React Native platform + platform version**: Android 5.0, Android 6.0.1

### react-native-camera
**Version**: 0.40.0

weremax, 2016-12-12:
While testing it has been revealed that the Camera does not like being embedded into additional Views.

grahamp, 2016-12-14:
I see this too, it seems to happen when the app is put into the background.

Batistleman, 2016-12-15:
I'm getting the same problem. The preview is sometimes empty, sometimes black. 

When I set `torchMode={Camera.constants.TorchMode.on}`, nothing happens.

Can I somehow get more debug info to help?

cooperka, 2016-12-16:
I got the same "white screen" issue on Android. This library is using the deprecated [Camera class](https://developer.android.com/reference/android/hardware/Camera.html), instead of the newer [camera2 API](https://developer.android.com/reference/android/hardware/camera2/package-summary.html), so there might be compatibility issues on newer devices. There were a few things I had to do to fix it:

1. Make sure you [declare permissions properly](https://developer.android.com/reference/android/hardware/Camera.html) in `AndroidManifest.xml`:

    ```xml
    <uses-permission android:name="android.permission.CAMERA" />
    <uses-feature android:name="android.hardware.camera" />
    <uses-feature android:name="android.hardware.camera.autofocus" />
    ```

2. Make sure you've given Camera permissions to your app. You can go to **Settings > Apps > [Your App] > Permissions** and enable the **Camera** setting.

3. It has never worked for me on an Android emulator, even with the camera enabled. It's a white screen still on the emulator, but it works just fine on my real device now (Nexus 6P with Android M).

grahamp, 2016-12-19:
@cooperka I agree you need to add the permissions as you outline above or you will get a blank screen 100% of the time. 
But there is another issue I am seeing consistently:
Try minimizing your app opening another app and resuming it.
Look at the logs. There are log messages showing a problem resuming the camera and you will get a blank screen.
12-14 00:01:57.767 459-2967/? E/Camera2Client: autoFocus: Camera 0: Call autoFocus when preview is inactive (state = 2).
12-14 00:01:58.767 459-2966/? E/Camera2Client: autoFocus: Camera 0: Call autoFocus when preview is inactive (state = 2).
12-14 00:01:59.767 459-459/? E/Camera2Client: autoFocus: Camera 0: Call autoFocus when preview is inactive (state = 2).
12-14 00:02:00.767 459-3458/? E/Camera2Client: autoFocus: Camera 0: Call autoFocus when preview is inactive (state = 2).
12-14 00:02:01.767 459-3453/? E/Camera2Client: autoFocus: Camera 0: Call autoFocus when preview is inactive (state = 2).
12-14 00:02:01.867 23627-23627/? D/Camera: [seungmin]_open start
12-14 00:02:01.867 23627-23627/? D/Camera: [China_security]_before
12-14 00:02:01.867 23627-23627/? D/Camera: [China_security]_after
12-14 00:02:01.867 23627-23627/? D/Camera: [seungmin]_open End
12-14 00:02:01.867 459-459/? D/CameraService: CameraService::connect E (pid 23627 "com.nauto.installerapp", id 0)
12-14 00:02:01.867 459-459/? W/CameraService: CameraService::connect X (pid 23627) rejected (existing client).
12-14 00:02:01.867 23627-23627/? W/CameraBase: An error occurred while connecting to camera: 0
12-14 00:02:01.867 23627-23627/? E/RCTCamera: acquireCameraInstance failed
                                              java.lang.RuntimeException: Fail to connect to camera service
                                                  at android.hardware.Camera.<init>(Camera.java:570)
                                                  at android.hardware.Camera.open(Camera.java:393)
                                                  at com.lwansbrough.RCTCamera.RCTCamera.acquireCameraInstance(RCTCamera.java:40)
                                                  at com.lwansbrough.RCTCamera.RCTCamera.<init>(RCTCamera.java:415)
                                                  at com.lwansbrough.RCTCamera.RCTCamera.createInstance(RCTCamera.java:33)
                                                  at com.lwansbrough.RCTCamera.RCTCameraView.<init>(RCTCameraView.java:30)
                                                  at com.lwansbrough.RCTCamera.RCTCameraViewManager.createViewInstance(RCTCameraViewManager.java:22)
                                                  at com.lwansbrough.RCTCamera.RCTCameraViewManager.createViewInstance(RCTCameraViewManager.java:12)
                                                  at com.facebook.react.uimanager.ViewManager.createView(ViewManager.java:46)
                                                  at com.facebook.react.uimanager.NativeViewHierarchyManager.createView(NativeViewHierarchyManager.java:218)
                                                  at com.facebook.react.uimanager.UIViewOperationQueue$CreateViewOperation.execute(UIViewOperationQueue.java:148)
                                                  at com.facebook.react.uimanager.UIViewOperationQueue$2.run(UIViewOperationQueue.java:776)
                                                  at com.facebook.react.uimanager.UIViewOperationQueue.flushPendingBatches(UIViewOperationQueue.java:829)
                                                  at com.facebook.react.uimanager.UIViewOperationQueue.access$1500(UIViewOperationQueue.java:44)
                                                  at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded(UIViewOperationQueue.java:868)
                                                  at com.facebook.react.uimanager.GuardedChoreographerFrameCallback.doFrame(GuardedChoreographerFrameCallback.java:32)
                                                  at com.facebook.react.uimanager.ReactChoreographer$ReactChoreographerDispatcher.doFrame(ReactChoreographer.java:131)
                                                  at android.view.Choreographer$CallbackRecord.run(Choreographer.java:765)
                                                  at android.view.Choreographer.doCallbacks(Choreographer.java:580)
                                                  at android.view.Choreographer.doFrame(Choreographer.java:549)
                                                  at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:753)
                                                  at android.os.Handler.handleCallback(Handler.java:739)
                                                  at android.os.Handler.dispatchMessage(Handler.java:95)
                                                  at android.os.Looper.loop(Looper.java:135)
                                                  at android.app.ActivityThread.main(ActivityThread.java:5431)
                                                  at java.lang.reflect.Method.invoke(Native Method)
                                                  at java.lang.reflect.Method.invoke(Method.java:372)
                                                  at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:914)
                                                  at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:707)
12-14 00:02:01.877 23627-23627/? D/Camera: [seungmin]_open start
12-14 00:02:01.877 23627-23627/? D/Camera: [China_security]_before
12-14 00:02:01.877 23627-23627/? D/Camera: [China_security]_after
12-14 00:02:01.877 23627-23627/? D/Camera: [seungmin]_open End
12-14 00:02:01.877 459-2967/? D/CameraService: CameraService::connect E (pid 23627 "com.nauto.installerapp", id 1)

luis-keego-io, 2017-03-09:
I got the same problem with physical devices with 4.2.2 and 4.3 android version.

- I added the permission
`<uses-permission android:name="android.permission.CAMERA" />`
`<uses-feature android:name="android.hardware.camera" />`
`<uses-feature android:name="android.hardware.camera.autofocus" />`

- I thought that the problem was due to [Camera class](https://developer.android.com/reference/android/hardware/Camera.html), but [Camera2 api](https://developer.android.com/reference/android/hardware/camera2/package-summary.html) was added in api level 21

- The exception thrown is **Fail to connect to camera service**

- The error is in the line `Camera camera = Camera.open(_cameraTypeToIndex.get(type));`

- the Logcat return 

> /RCTCamera: acquireCameraInstance failed
>                                                   java.lang.RuntimeException: Fail to connect to camera service
>                                                       at android.hardware.Camera.native_setup(Native Method)
>                                                       at android.hardware.Camera.<init>(Camera.java:410)
>                                                       at android.hardware.Camera.open(Camera.java:365)
>                                                       at com.lwansbrough.RCTCamera.RCTCamera.acquireCameraInstance(RCTCamera.java:43)
>                                                       at com.lwansbrough.RCTCamera.RCTCamera.<init>(RCTCamera.java:442)
>                                                       at com.lwansbrough.RCTCamera.RCTCamera.createInstance(RCTCamera.java:34)
>                                                       at com.lwansbrough.RCTCamera.RCTCameraView.<init>(RCTCameraView.java:30)
>                                                       at com.lwansbrough.RCTCamera.RCTCameraViewManager.createViewInstance(RCTCameraViewManager.java:22)
>                                                       at com.lwansbrough.RCTCamera.RCTCameraViewManager.createViewInstance(RCTCameraViewManager.java:12)
>                                                       at com.facebook.react.uimanager.ViewManager.createView(ViewManager.java:46)
>                                                       at com.facebook.react.uimanager.NativeViewHierarchyManager.createView(NativeViewHierarchyManager.java:218)
>                                                       at com.facebook.react.uimanager.UIViewOperationQueue$CreateViewOperation.execute(UIViewOperationQueue.java:148)
>                                                       at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.dispatchPendingNonBatchedOperations(UIViewOperationQueue.java:890)
>                                                       at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded(UIViewOperationQueue.java:863)
>                                                       at com.facebook.react.uimanager.GuardedChoreographerFrameCallback.doFrame(GuardedChoreographerFrameCallback.java:32)
>                                                       at com.facebook.react.uimanager.ReactChoreographer$ReactChoreographerDispatcher.doFrame(ReactChoreographer.java:131)
>                                                       at android.view.Choreographer$CallbackRecord.run(Choreographer.java:789)
>                                                       at android.view.Choreographer.doCallbacks(Choreographer.java:591)
>                                                       at android.view.Choreographer.doFrame(Choreographer.java:560)
>                                                       at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:777)
>                                                       at android.os.Handler.handleCallback(Handler.java:725)
>                                                       at android.os.Handler.dispatchMessage(Handler.java:92)
>                                                       at android.os.Looper.loop(Looper.java:176)
>                                                       at android.app.ActivityThread.main(ActivityThread.java:5365)
>                                                       at java.lang.reflect.Method.invokeNative(Native Method)
>                                                       at java.lang.reflect.Method.invoke(Method.java:511)
>                                                       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1102)
>                                                       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:869)
>                                                       at dalvik.system.NativeStart.main(Native Method)

Sorry for my broken english!

knapeto, 2017-03-30:
Any solution pls?

seppemarotta, 2017-04-03:
I'm having this issue, i have my camera embed in a View and it's empty too .

knapeto, 2017-04-04:
It resolved all of this crashes on `npm i react-native-camera@0.6`

KanUchida, 2017-05-05:
I'm using react-native-camera 0.6 with react native 0.43.3 but I still got this error

guerjon, 2017-05-16:
I update to react-native-camera@0.6 but still not work @knapeto, same error

acquireCameraInstance failed
java.lang.RuntimeException: Fail to connect to camera service

Batistleman, 2017-05-16:
Please make sure you have allowed the program to use the camera (applications-> app -> permissions)

weremax, 2017-08-04:
closing this issue

closed 2017-08-04 by weremax

KanUchida, 2017-08-12:
I figure out why this happen in my case.

When you take camera, you can't update your props.

I was using react-native with redux to manage state well. And I dispatch action and update props when camera is taken. This is where this problem happens.

To avoid this, I use ```this.setState({ ... })``` instead of update props with ```this.props.dospatchActionToUpdateProps()```. Then everything works well!

Hope this will help someone.

fabriziobertoglio1987, 2019-01-10:
as suggested a debugged the view and found out, as displayed in the picture that the `Camera` component `state.isAuthorized: false`. I switched to true and it works.

[image]

darion6, 2022-08-09:
how did you resolve the white screen 
 issue

darion6, 2022-08-09:
I solve it by using flex: in the style and adding permission in manifest.