react-native-camera/react-native-camera #515: Getting Error in Build regarding Permission last Commit

joypatel04, 2016-12-14:
### Actual behaviour
Getting Error in Build 

[image]

Looks like something to do with Permission.
But Since i am on react-native 0.34 i don't know how i can resolve it

### Environment
- **Node.js version**: v7.2.1
- **React Native version**:  0.34.1
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version**:  "master"

iandea, 2016-12-14:
Seeing the same here.  RN Version 0.38

joypatel04, 2016-12-14:
Looks like we have to use http://facebook.github.io/react-native/docs/permissionsandroid.html 
@iandea

charpeni, 2016-12-14:
Sorry, it should be fixed now.

closed 2016-12-14 by charpeni

joypatel04, 2016-12-14:
Thanks @charpeni

PR #516 (2016-12-14): Revert "Add video/audio authorization check for Android"
Reverts lwansbrough/react-native-camera#513

Fixes #515