react-native-camera/react-native-camera #518: react-native-camera front camera not working on ios

kennedc, 2016-12-15:
Front camera only works sometimes on ios, it seems that it works on the first time the app is opened but then stops? if i reload the js, then it works again. works perfectly on Android

Function to swtich camera:

```
switchType() {
  let newType;
  const { back, front } = Camera.constants.Type;

  if (this.state.camera.type === back) {
    newType = front;
  } else if (this.state.camera.type === front) {
    newType = back;
  }

  this.setState({
    camera: {
      ...this.state.camera,
      type: newType,
    },
  });
}
```
Markup:

```
<Camera
            ref={(cam) => {
              this.camera = cam;
            }}
            style={styles.preview}
            aspect={this.state.camera.aspect}
            captureTarget={this.state.camera.captureTarget}
            type={this.state.camera.type}
            flashMode={this.state.camera.flashMode}
            defaultTouchToFocus
            mirrorImage={false}
            captureQuality="medium"
          />
```

kennedc, 2016-12-15:
This happens when using the Navigator, when you push a Route and the pop back to the camera route. Ive seen there is some unmounting problems. Does anyone have a fix for this? on first load after rebooting the phone it works.

ju5td0m7m1nd, 2017-01-07:
Same Issue Here

kennedc, 2017-01-07:
I solved it by using this.props.navigator.replace() instead of push and pop, but i like your solution!

ospfranco, 2018-01-13:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-13 by ospfranco