react-native-camera/react-native-camera #519: Can I take a photo while recording a video on Android?

MedusaLeee, 2016-12-16:
I modify the demo code to support  can take a photo while in recording a video(can click capture photo Button while in  recording);

I test it success on Ios, but failed on Android;

I want to take a photo while in recording a video on Android， what  can i do?

My English is not good, I hope you can understand; Thanks;

 The Example.js :

```javascript
export default class Example extends React.Component {
	constructor(props) {
		super(props);

		this.camera = null;

		this.state = {
			camera: {
				aspect: Camera.constants.Aspect.fill,
				captureTarget: Camera.constants.CaptureTarget.cameraRoll,
				type: Camera.constants.Type.back,
				orientation: Camera.constants.Orientation.auto,
				flashMode: Camera.constants.FlashMode.auto
			},
			isRecording: false
		};

		this.takePicture = this.takePicture.bind(this);
		this.startRecording = this.startRecording.bind(this);
		this.stopRecording = this.stopRecording.bind(this);
		this.switchType = this.switchType.bind(this);
		this.switchFlash = this.switchFlash.bind(this);
	}

	takePicture() {
		if (this.camera) {
			this.camera.capture()
				.then((data) => {
				})
				.catch(err => alert('takePicture'+err));
		}
	}

	startRecording() {
		if (this.camera) {
			this.camera.capture({mode: Camera.constants.CaptureMode.video})
				.then((data) => console.log(data))
				.catch(err => console.error(err));
			this.setState({
				isRecording: true
			});
		}
	}

	stopRecording() {
		if (this.camera) {
			this.camera.stopCapture();
			this.setState({
				isRecording: false
			});
		}
	}

	switchType() {
		let newType;
		const {back, front} = Camera.constants.Type;

		if (this.state.camera.type === back) {
			newType = front;
		} else if (this.state.camera.type === front) {
			newType = back;
		}

		this.setState({
			camera: {
				...this.state.camera,
				type: newType,
			},
		});
	}

	get typeIcon() {
		let icon;
		const {back, front} = Camera.constants.Type;

		if (this.state.camera.type === back) {
			icon = require('./assets/ic_camera_rear_white.png');
		} else if (this.state.camera.type === front) {
			icon = require('./assets/ic_camera_front_white.png');
		}

		return icon;
	}

	switchFlash() {
		let newFlashMode;
		const {auto, on, off} = Camera.constants.FlashMode;

		if (this.state.camera.flashMode === auto) {
			newFlashMode = on;
		} else if (this.state.camera.flashMode === on) {
			newFlashMode = off;
		} else if (this.state.camera.flashMode === off) {
			newFlashMode = auto;
		}

		this.setState({
			camera: {
				...this.state.camera,
				flashMode: newFlashMode,
			},
		});
	}

	get flashIcon() {
		let icon;
		const {auto, on, off} = Camera.constants.FlashMode;

		if (this.state.camera.flashMode === auto) {
			icon = require('./assets/ic_flash_auto_white.png');
		} else if (this.state.camera.flashMode === on) {
			icon = require('./assets/ic_flash_on_white.png');
		} else if (this.state.camera.flashMode === off) {
			icon = require('./assets/ic_flash_off_white.png');
		}

		return icon;
	}

	render() {
		return (
			<View style={styles.container}>
				<StatusBar
					animated
					hidden
				/>
				<Camera
					ref={(cam) => {
			            this.camera = cam;
			          }}
					style={styles.preview}
					aspect={this.state.camera.aspect}
					captureTarget={this.state.camera.captureTarget}
					type={this.state.camera.type}
					flashMode={this.state.camera.flashMode}
					defaultTouchToFocus
					mirrorImage={false}
				/>
				<View style={[styles.overlay, styles.topOverlay]}>
					<TouchableOpacity
						style={styles.typeButton}
						onPress={this.switchType}
					>
						<Image
							source={this.typeIcon}
						/>
					</TouchableOpacity>
					<TouchableOpacity
						style={styles.flashButton}
						onPress={this.switchFlash}
					>
						<Image
							source={this.flashIcon}
						/>
					</TouchableOpacity>
				</View>
				<View style={[styles.overlay, styles.bottomOverlay]}>
                                       // I modify this, I can click capture photo Button while recording ;
					{
						<TouchableOpacity
							style={styles.captureButton}
							onPress={this.takePicture}
						>
							<Image
								source={require('./assets/ic_photo_camera_36pt.png')}
							/>
						</TouchableOpacity>
					}
					<View style={styles.buttonsSpace}/>
					{
						!this.state.isRecording
						&&
						<TouchableOpacity
							style={styles.captureButton}
							onPress={this.startRecording}
						>
							<Image
								source={require('./assets/ic_videocam_36pt.png')}
							/>
						</TouchableOpacity>
						||
						<TouchableOpacity
							style={styles.captureButton}
							onPress={this.stopRecording}
						>
							<Image
								source={require('./assets/ic_stop_36pt.png')}
							/>
						</TouchableOpacity>
					}
				</View>
			</View>
		);
	}
}
```

ospfranco, 2018-01-13:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-13 by ospfranco

aurazaharia, 2018-04-02:
Hello,

Did you manage to make this work?

sibelius, 2018-04-02:
Open another issue if this still valid on RNCamera

kamleshtilwani, 2018-07-13:
Hey! 
Were you successful to implement the same??

theothersideofgod, 2020-03-02:
is this still a bug? I still can't take picture while recording on android but ios can