react-native-camera/react-native-camera #522: Getting "Attempt to invoke virtual method ... on a null object reference" error while capturing

peterKaleta, 2016-12-19:
While capturing a photo on `React Native 0.39.1` / `Android 7`
 my component/app crashes with the error listed below

### Code

```
  takePicture() {

    this.camera.capture()
    .then((data) => {

      const image = {
        uri: data.path,
        height: 640,
        width: 480
      };

      ExpenseActions.updateValues({key: 'photo', value: image});
      this.context.navigator.forward('ConfirmPhotoToExpense');
    })
    .catch(err => console.error(err));
  }

...
 render() {

    const cameraIconSize = 22;
    const cameraIconName = 'camera';

    return (
      <View style={cameraStyles.container}>
        <Camera
          ref={(cam) => {this.camera = cam;}}
          style={cameraStyles.preview}
          captureTarget={Camera.constants.CaptureTarget.temp}
          orientation="portrait">

          <View style={cameraStyles.capture}>
            <View style={[styles.icon, cameraStyles.cameraIcon]}>
              <Icon
                name={cameraIconName}
                color={WHITE} size={cameraIconSize}
                onPress={() => this.takePicture()}
              />
            </View>
          </View>
        </Camera>
        <View style={cameraStyles.photoGrid}>
          {this.state.photos.map(this.renderImage)}
        </View>
      </View>
    );
  }
```

### Error
Throws this beauty 
```
/AndroidRuntime: FATAL EXCEPTION: main
                                                                  Process: com.awesomeproject, PID: 2962
                                                                  java.lang.NullPointerException: Attempt to invoke virtual method 'boolean com.drew.metadata.Directory.containsTag(int)' on a null object reference
                                                                      at com.lwansbrough.RCTCamera.RCTCameraModule.fixOrientation(RCTCameraModule.java:576)
                                                                      at com.lwansbrough.RCTCamera.RCTCameraModule.access$300(RCTCameraModule.java:45)
                                                                      at com.lwansbrough.RCTCamera.RCTCameraModule$3.onPictureTaken(RCTCameraModule.java:668)
                                                                      at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1092)
                                                                      at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                      at android.os.Looper.loop(Looper.java:154)
                                                                      at android.app.ActivityThread.main(ActivityThread.java:6119)
                                                                      at java.lang.reflect.Method.invoke(Native Method)
                                                                      at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:886)
                                                                      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:776)
12-18 18:34:08.874 1563-1902/system_process W/ActivityManager:   Force finishing activity com.awesomeproject/.MainActivity
Chat Conversation End
```

amanthegreatone, 2016-12-27:
Same issue with me. Making dummy apps for component functionality before starting building the main app. 

Code:

/**
 * Sample React Native App
 * https://github.com/facebook/react-native
 * @flow
 */

import React, { Component } from 'react';
import {
  StyleSheet,
  Text,
  View
} from 'react-native';

import Camera from 'react-native-camera';
import SimpleButton from './SimpleButton';

export default class CameraScreen extends Component {

  _takePicture () {
    this.refs.camera.capture( (err, data) => {
      if (err) return;
      this.props.onPicture(data);
    });
  }

  takePicture() {
    this.camera.capture()
      .then( (data) => {console.log(data); this.props.onPicture(data); } )
      .catch(err => console.error(err));
  }

  render() {
    return(
      <Camera
        captureTarget={Camera.constants.CaptureTarget.disk}
        //ref="cam"
        ref={(cam) => {
            this.camera = cam;
          }}
        style={styles.container}
        aspect={Camera.constants.Aspect.fill}
        orientation="portrait"
        >
        <View style={styles.cameraButtonContainer}>
          <SimpleButton
            onPress={this.takePicture.bind(this)}
            customText="Capture"
            style={styles.cameraButton}
            textStyle={styles.cameraButtonText}
          />
        </View>
      </Camera>
    );
  }

}

var styles = StyleSheet.create({
  container: {
    flex: 1,
    marginTop: 64
  },
  cameraButtonContainer: {
    position: 'absolute',
    bottom: 20,
    left: 20,
    right: 20
  },
  cameraButton: {
    backgroundColor: '#5B29C1',
    borderRadius: 4,
    paddingHorizontal: 20,
    paddingVertical: 15
  },
  cameraButtonText: {
    color: 'white',
    textAlign: 'center'
  }
});

Here is the error log:

2-27 14:17:19.631 11788-11788/com.reactnotes E/AndroidRuntime: FATAL EXCEPTION: main
                                                                java.lang.NullPointerException
                                                                    at com.lwansbrough.RCTCamera.RCTCameraModule.fixOrientation(RCTCameraModule.java:576)
                                                                    at com.lwansbrough.RCTCamera.RCTCameraModule.access$300(RCTCameraModule.java:45)
                                                                    at com.lwansbrough.RCTCamera.RCTCameraModule$3.onPictureTaken(RCTCameraModule.java:668)
                                                                    at android.hardware.Camera$EventHandler.handleMessage(Camera.java:734)
                                                                    at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                    at android.os.Looper.loop(Looper.java:137)
                                                                    at android.app.ActivityThread.main(ActivityThread.java:4745)
                                                                    at java.lang.reflect.Method.invokeNative(Native Method)
                                                                    at java.lang.reflect.Method.invoke(Method.java:511)
                                                                    at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:786)
                                                                    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:553)
                                                                    at dalvik.system.NativeStart.main(Native Method)

wardnath, 2017-01-01:
+1 getting the exact same issue with the Example on master

wardnath, 2017-01-01:
so I did a factory reset on simulator then ran the example from master again, I got an issue with capture but this time the error was related to `libwvm.so` not being found. I set the parameter `playSoundOnCapture` to false. The Example project is now working when I have a fresh copy of master.

wardnath, 2017-01-02:
looks like this issue might be related: https://github.com/lwansbrough/react-native-camera/issues/530

swaroopa94, 2017-08-03:
Hi,
I got the same issue in the real device as well 
Can you Please tell me how to solve this as it is urgent

capozzid, 2017-08-31:
I have the same error, RN0.47.2 Camera 0.10.0

gustaver, 2017-09-13:
Got the same issue on real device as well. RN 0.40. The reason I'm getting it is because I'm automatically calling `this.camera.capture()` when the Camera mounts, and on Android this causes a crash. Would really appreciate some workaround or way to fix this.

ospfranco, 2018-01-13:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

tsubery, 2018-01-19:
I saw the same error on latest stable release 0.12.
```
java.lang.NullPointerExceptionNavigationActivity
Attempt to invoke virtual method 'android.hardware.Camera$Parameters android.hardware.Camera.getParameters()' on a null object reference
```
this is the top of the stack trace:

```
RCTCameraViewFinder.java:363com.lwansbrough.RCTCamera.RCTCameraViewFinder.onTouchEvent
View.java:9993 android.view.View.dispatchTouchEvent	
ViewGroup.java:2839 android.view.ViewGroup.dispatchTransformedTouchEvent	
ViewGroup.java:2456 android.view.ViewGroup.dispatchTouchEvent
```

Anyone have clues about the cause, solution or possible workaround?
Thanks!

sibelius, 2018-01-19:
for some weird reason

the null check is not working well here

https://github.com/react-native-community/react-native-camera/blob/a4ad0446cad244b2f79384bb7e22730062efe17e/android/src/main/java/com/lwansbrough/RCTCamera/RCTCameraViewFinder.java#L396

tsubery, 2018-01-19:
This null check was introduced after the latest stable release.
I'll just branch off 0.12 for now.
Thanks!

AyoubXwire, 2019-09-16:
I got the same problem after changing a class component to functional component... the solution was the just uninstall the app and reinstall it using the command `react-native run-android` on android

mayu-496, 2019-11-22:
I was getting the same problem with expo .. I just stopped  running the app ,  cleared expo recently open project and again started with ' expo start ' ... This worked for me..

MaruAfrica, 2020-02-10:
Awesome @AyoubXwire  I also manage to solve it on expo the fix was just to clear data and cache on the expo client app that's it !

twero, 2020-06-11:
I have the same issue,
still cant figure it out

closed 2020-08-18 by stale[bot]

johnilizy17, 2021-09-27:
I solve this problem by starting a new react native project because it save me all the stress of looking for unknown solutions