react-native-camera/react-native-camera #527: Rotating the image in memory can lead to Out of Memory exception

gmahomarf, 2016-12-21:
### Steps to reproduce
1. Call `capture()` in picture mode with quality set to high

### Expected behaviour
The app should not crash

### Actual behaviour
Sometimes, the app throws an Out of Memory Exception

### Environment
- **Node.js version**: 6.9.1
- **React Native version**: 0.32
- **React Native platform + platform version**: Android 6.0

### react-native-camera
**Version**: master

After some debugging: the OOM Exception happens during the call to [fixOrientation](https://github.com/lwansbrough/react-native-camera/blob/master/android/src/main/java/com/lwansbrough/RCTCamera/RCTCameraModule.java#L568-L585). Even when it doesn't crash, it takes a while for capture to end. Wouldn't it be more efficient to change the orientation when the file has already been written, like [rewriteOrientation](https://github.com/lwansbrough/react-native-camera/blob/master/android/src/main/java/com/lwansbrough/RCTCamera/RCTCameraModule.java#L587-L595) does?

npomfret, 2017-03-21:
You can add the `largeHeap` setting to your manifest file:

```
<application
        android:name=".MainApplication"
        android:largeHeap="true"
```

Also, my _performance_ fork has some android performance and memory fixes in - not submitted a PR yet, still testing. https://github.com/npomfret/react-native-camera/tree/performance

npomfret, 2017-04-26:
@gmahomarf are you able to test my PR?  It may help with this https://github.com/lwansbrough/react-native-camera/pull/682.

skovhus, 2017-08-18:
@npomfret I've tested out #682 (0.10.0):
- you can now avoid the memory crash if not using the `fixOrientation` flag
- but the rotation is still wrong -90 for front facing camera the different Android devices I tested (Samsung Galaxy S7 / Android 7 and a Samsung Galaxy S4 Android 5)

ospfranco, 2018-01-13:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-13 by ospfranco