react-native-camera/react-native-camera #528: When I try to run Example on IOS I get: undefined is not an object (evaluating 'CameraManager.Aspect')

idvorkin, 2016-12-21:
## Steps to reproduce
1. Build example app (include change bundle name so can sign it)
2. Run
3.

### Expected behaviour
It should run the example 

### Actual behaviour
It got the critical error:  undefined is not an object (evaluating 'CameraManager.Aspect')
### Environment
- **Node.js version**: v7.2.0
- **React Native version**: 0.34.1
- **React Native platform + platform version**: iOS 10.2
### react-native-camera
**Version**: master

GuillaumeSarfati, 2016-12-22:
Yes me too!  just when i import the react-native-camera plugin

ecavalcanti, 2016-12-23:
Run: react-native link react-native-camera

freethejazz, 2016-12-24:
There's an existing issue documenting this. Not sure why it's closed, but it should be helpful: https://github.com/lwansbrough/react-native-camera/issues/272

Erin007, 2017-01-11:
Any progress or suggestions for this issue? Getting the same error with React Native .4

sashakid, 2017-01-31:
`react-native link react-native-camera` helped me

anonrig, 2017-02-14:
Even though i did react-native link still have this issue. Any solutions?

Marthyn, 2017-02-28:
Remove your build folder and run `react-native link react-native-camera` again!

gwenf, 2017-04-03:
I had to restart the RN Packager and emulator after I ran `react-native link react-native-camera`.
Then it started working.

kylanhurt, 2017-04-17:
I am still having this issue. @Marthyn How do I find which folder is being used as the build folder? I'm not seeing anything in my regular project folder.

encorex, 2017-04-23:
same issue...

eqyiel, 2017-05-05:
Maybe related #692

Rainer2465, 2017-05-09:
1. try npm link react-native-camera again
2. delete node_modules and npm install | npm upgrade - npm install
3. don't start the app with npm start, use react-native run-ios | react-native run-ios --simulator "iPhone 7" (for a especific device)

enrique26, 2017-09-06:
I dont know exactly if this gonna help, but i recently resolved this issue by doing this:
-open the xcodeproj 
-on build phases ,  add the "linRCTcamera" binary manually (on my case, even if you used "react-native link" command, some binaries are not added to the project correctly)
-close the project and finally try run again the project in the terminal

i did notice this after many tries , deleting and recreating the project many times
hope help someone

ospfranco, 2018-01-13:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-13 by ospfranco