react-native-camera/react-native-camera #530: Android orientation fix crash on Samsung phone

radko93, 2016-12-22:
### Steps to reproduce
1. Same like here #475 

### Expected behaviour
After fix in #493, it should save a correct image.

### Actual behaviour
App Crashes. Seems that there is no ExifIFD0Directory in the image, so you can't read it [here](https://github.com/lwansbrough/react-native-camera/blob/master/android/src/main/java/com/lwansbrough/RCTCamera/RCTCameraModule.java#L575). Probably the problem is that the third party library fails to read the exif: https://github.com/drewnoakes/metadata-extractor/issues/105.
@alwx you might be interested in this :)

### Environment
- **Node.js version**: `7.2`
- **React Native version**: `0.39.2`
- **React Native platform + platform version**: Galaxy S7, Android 6.0

### react-native-camera
**Version**: `master`

wardnath, 2017-01-02:
Perhaps there could just be a check if the metadata exists, if not then just return the unaltered data

```

    private byte[] fixOrientation(byte[] data) {
        final Metadata metadata;
        try {
            metadata = ImageMetadataReader.readMetadata(
                    new BufferedInputStream(new ByteArrayInputStream(data)), data.length
            );

            final ExifIFD0Directory exifIFD0Directory = metadata.getFirstDirectoryOfType(ExifIFD0Directory.class);
            // return unaltered data if exifIFD0Directory is null 
            if (exifIFD0Directory == null) {
              return data;
            } else if (exifIFD0Directory.containsTag(ExifIFD0Directory.TAG_ORIENTATION)) {
                final int exifOrientation = exifIFD0Directory.getInt(ExifIFD0Directory.TAG_ORIENTATION);
                return rotate(data, exifOrientation);
            }
            return data;
        } catch (IOException | ImageProcessingException | MetadataException e) {
            e.printStackTrace();
            return data;
        }
    }
```

npomfret, 2017-04-26:
@radko93 are you able to test my PR?  it may help with this https://github.com/lwansbrough/react-native-camera/pull/682

ospfranco, 2018-01-13:
I see there is an open PR, I will close the ticket and review the PR later

closed 2018-01-13 by ospfranco