react-native-camera/react-native-camera #533: Slow Motion Video

Kottidev, 2016-12-27:
### Steps to reproduce
1. Slow Motion Video

### Expected behaviour

Video should be Recorded slowly or Speedly ( can you ad new param for example timescale

// timescale = Higher than 1 makes a slow motion, between 0 and 1 makes a timelapse effect

### Actual behaviour
i don't think you have this fonctionality 

### Environment
- **Node.js v6.7.0**:
- **React Native v0.37.0**:
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version**: npm version or "master"

banandelfiner, 2017-03-24:
Is there anyone working on a function for this?

banandelfiner, 2017-06-06:
I have implemented a function in iOS for this (in my fork). It's not set in props though. You just call this.camera.setFrameRate(120), as an example.

I guess your are talking about higher frame rate?

RekkyRek, 2017-07-05:
What i would do is set the framerate of the camera higher (only works on phones with support for higher framerate) and then use some video processing package to slow it down after capture.

ospfranco, 2018-01-13:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-13 by ospfranco

ankitdetroja, 2021-09-02:
@leahlundqvist 
Hey! can you help me on this?if yes,  it would be great.