react-native-camera/react-native-camera #539: App stops when I try to capture an image.

ansierras, 2017-01-03:
### Steps to reproduce
1. Open Camera App
2. Try to capture image
3.

### Expected behaviour
The app should take a picture and console log the result.

### Actual behaviour
The app stops (not crashes) at a thread stop that says" __abort_with_payload".
I try to continue program execution, but it still stops.
I have tried using low quality picture capture, but the problem persists. 
### Environment
- **Node.js version**: latest
- **React Native version**: latest
- **React Native platform + platform version**: iOS

### react-native-camera
**Version**: npm version

hugohow, 2017-01-04:
I have the same issue !

ansierras, 2017-01-06:
Can you tell me how you solved it?

hugohow, 2017-01-06:
In my Info.plist, I added this, and it works : 

```
<key>NSCameraUsageDescription</key>
<string>Your message to user when the camera is accessed for the first time</string>

<!-- Include this only if you are planning to use the camera roll -->
<key>NSPhotoLibraryUsageDescription</key>
<string>Your message to user when the photo library is accessed for the first time</string>

<!-- Include this only if you are planning to use the microphone for video recording -->
<key>NSMicrophoneUsageDescription</key>
<string>Your message to user when the microsphone is accessed for the first time</string>
```

ansierras, 2017-01-06:
This doesn't solve the issue for me :(

RikNieuwoudt, 2017-01-11:
I'm also having this issue.

hudjoubert, 2017-01-24:
This info.plist is only for iphone or i have to do this on android too?

julesmoretti, 2017-02-06:
I had the same issue on iOS until I allowed access to camera roll.

```
  <!-- Include this only if you are planning to use the camera roll -->
  <key>NSPhotoLibraryUsageDescription</key>
  <string>Your message to user when the photo library is accessed for the first time</string>
```

ansierras, 2017-02-10:
Thank you @julesmoretti ! That solved my issue. I was only accessing the camera in the info.plist, not the camera roll. I added access to the camera roll and now it works perfectly. 
Thanks a lot!

closed 2017-02-10 by ansierras