react-native-camera/react-native-camera #551: libsystem_kernel.dylib`__pthread_kill:

chipe13, 2017-01-08:
### Steps to reproduce
1. Installed react native 0.40
2. npm i react-native-camera@next 
    -followed steps in manual setup for xcode
3. followed code in readme

### Expected behaviour
content renders on page and allows you to click on [CAPTURE]

### Actual behaviour
Builds successful but after simulator opens xcode goes to a new screen with:

libsystem_kernel.dylib`__pthread_kill:
    0x1077d8efc <+0>:  movl   $0x2000148, %eax          ; imm = 0x2000148 
    0x1077d8f01 <+5>:  movq   %rcx, %r10
    0x1077d8f04 <+8>:  syscall 
->  0x1077d8f06 <+10>: jae    0x1077d8f10               ; <+20>
    0x1077d8f08 <+12>: movq   %rax, %rdi
    0x1077d8f0b <+15>: jmp    0x1077d37cd               ; cerror_nocancel
    0x1077d8f10 <+20>: retq   
    0x1077d8f11 <+21>: nop    
    0x1077d8f12 <+22>: nop    
    0x1077d8f13 <+23>: nop    

### Environment
- **Node.js version**:
- **React Native version**:
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
"^1.0.0-alpha1"

I cant seem to get this working right out of the box following the README. What am I forgetting or whats happening with this issue?

chipe13, 2017-01-09:
Sorry. Ignore. I missed the step of code I had to add if on ios 10+

Please Close

closed 2017-02-17 by ptomasroos

kodatech, 2017-08-08:
Sorry @chipe13, where is this step?

kodatech, 2017-08-08:
I've already found it. Thanks @chipe13

louguoquan, 2018-03-26:
I have the same problem.  @chipe13, where is this step?