react-native-camera/react-native-camera #552: No audio record With Video for  ios 10.2

nbappi, 2017-01-09:
### Steps to reproduce
1. Installed react native 0.34
2. npm i react-native-camera@next
     -followed steps in manual setup for xcode
3. followed code in readme

### Expected behaviour
The apps should record video with audio.

### Actual behaviour
This app work very well to record video.

I have faced some problem : 
1. No audio record with video.

### Environment
- **Node.js version**:  6.9.3
- **React Native version**: 0.34.0
- **React Native platform + platform version**: iOS 10.2
### react-native-camera
**Version**: npm version or "master"

walbell, 2017-02-13:
Did you try to set captureAudio=true in the camera configuration?
Its default value is false.

ospfranco, 2018-01-13:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

P.S. There is a new implementation of the camera on android RNCamera, it is on the master branch, you can try it and see if it fixes your problem

closed 2018-01-13 by ospfranco