react-native-camera/react-native-camera #556: Camera run always (Display off, in Background etc)

fasibio, 2017-01-16:
### Steps to reproduce
Buid a app with this code in renderer
 ```
<Camera
        ref="cam"
        barCodeTypes={['ean13']}
        onBarCodeRead={(data,bounds ) =>{
          console.log('read',data, bounds)
        }}
        aspect={Camera.constants.Aspect.fill}
        style={styles.preview}
        flashMode={Camera.constants.FlashMode.on}
        />
```
Start react-native log-android 
Take off the display. 
Scan an ean-13 barcode. 
You can see a output in the log

You have to override the Lifecycle Methods... 

### Environment
- React native : 0.40.0:
- Platform Nexus 6P Android 7.1.1

### react-native-camera
"react-native-camera": "^1.0.0-alpha1"

juanluisgv1, 2017-01-25:
I had the same issue, have you unmounted the component which renders the camera?

carleryd, 2017-01-26:
I have had this issue for a long time and solved it as @juanluisgv1 suggests. This however makes switching from an to the camera view really slow.
Comparing a Nexus 5x device using this "hack" it takes over 1000ms to switch to and from the camera view, while an iPhone 5s doesn't need this hack and switches between camera and other view in about 50ms.
It seems that on iPhone, it switches to the camera view, and is initially black until the native camera has loaded. I would like to have the same behavior on Android

wenwen1995, 2017-08-21:
I  think  @carleryd  idea is a good solution,and I find another solution to solve this problem,Please look at this link:[https://www.thepolyglotdeveloper.com/2015/10/implement-a-barcode-scanner-using-react-native/](url),
it may set a flag as true in initial  to render the camera, if the flag equals false, return null...When the onBarCodeRead function works,it sets the flag false.That's will prevent the Camera always works.

`this.state = { barCodeFlag : true}`

` render() {
    return (
    	<View style={_styles.container}>
         {  this.renderCamera()  }
        </View>
    	)
    }`

`  renderCamera() {
       let { barCodeFlag } = this.state;
        if(barCodeFlag ) {
            return (
                <Camera
                    ref="cam"
                    style={styles.container}
                    onBarCodeRead={this.onBarCodeRead.bind(this)}
                    type={this.state.cameraType}>
                </Camera>
            );
        } else {
            return  null;
        }
    },`

`onBarCodeRead(e){
      this.setState({barCodeFlag: false}); 
      Vibration.vibrate(); //扫码震动
     console.log('e.data=>',e.data);
}
`

galhardi, 2017-09-09:
@wenwen1995, how can I enable the camera again?

wenwen1995, 2017-09-09:
I don't understand your thoughts...You main like this?First time the  camera reads a code data,and prints it,and next time,you want the camera to reads another code data,and prints this data again?

galhardi, 2017-09-09:
I need enable cam when navigate (using tab navigation) to a screen and disable when navigate to another screen.
In my case when I read code and redirect to another screen my can continue activate and read another codes.
Thks for your response.

milon87, 2017-09-10:
when it reads a code then  navigate to another acreen. Camera is still reading in the background!!!! If I stop reading then how to enable again when user backs to the camera screen ?

galhardi, 2017-09-10:
@milon87 exactly!

wenwen1995, 2017-09-11:
@galhardi ,may be you should use ** this.props.navigator.replace** to jump another screen,and this  will destory the current pages,so it can stop the camera working....Also,I think you should focus my first answer of this problem ,please set a flag as true in initially,when page  renders, it will start the camera, and when the onBarCodeRead function works,it sets the flag false.And you can jump another page after this.That's will prevent the Camera always works.

galhardi, 2017-09-11:
@wenwen1995, I tried use the flag. I set true on componentWillMount and false on  onBarCodeRead. But when I come back to camera page the componentWillMount not trigger again.
 How can I set true the flag again?
Thanks for your help.

wenwen1995, 2017-09-11:
[image]

I set flag true in initially,and this will be okay,not set it in the componentWillMount function,just have a try

galhardi, 2017-09-11:
@wenwen1995, I'm tried use constructor but it's called just in the first time that I enter on camera page.
I'm using react-navigation (TabNavigator), I don't know if this is the cause of my problem.

wenwen1995, 2017-09-11:
@galhardi ,oh I think it may be the react-navigation problem,when it backs to camera page,it doesn't trigger the render,So I think there is a way to solve it.---->You can jump  camera page 
with return parameters(about this ,you should search react-navigation about how to back a page with return parameters...  ),it will trigger the page render again,

ospfranco, 2018-01-13:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

P.S. There is a new implementation of the camera on android RNCamera, it is on the master branch, you can try it and see if it fixes your problem

closed 2018-01-13 by ospfranco

exhesham, 2018-02-05:
the answer is to put it in a state and then set that state to null - worked for me