react-native-camera/react-native-camera #557: Exception thrown from Xcode when barCodeTypes is set as an array

alexmngn, 2017-01-17:
### Steps to reproduce
1. Set the property:  `barCodeTypes={['ean13', 'ean8', 'upce']}`
2. Run the app

This is how my component looks like:
```
<Camera
	aspect={Camera.constants.Aspect.fill}
	style={styles.camera}
	torchMode={this.state.torch}
	onBarCodeRead={data => this.onBarCodeRead(data)}
	barCodeTypes={['ean13', 'ean8', 'upce']}
>
```
### Expected behaviour
Camera should launch

### Actual behaviour

Camera doesn't start, getting this exception:

```
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[AVCaptureMetadataOutput setMetadataObjectTypes:] Unsupported type found - use -availableMetadataObjectTypes'
*** First throw call stack:
(0x186e211b8 0x18585855c 0x18e662a48 0x1001f9a88 0x100431258 0x100431218 0x10043eaec 0x100434ce0 0x10043f088 0x100440e2c 0x100440b78 0x185eb32a0 0x185eb2d8c)
libc++abi.dylib: terminating with uncaught exception of type NSException
```

[image]

### Environment
- **Node.js version**: v6.0.0
- **React Native version**: 0.39
- **React Native platform + platform version**: iOS 10

### react-native-camera
**Version**: 0.4.0

jarredszabadi, 2017-01-24:
also experiencing this issue

alexmngn, 2017-01-26:
Found out looking at the code that the type must be coming from the `Camera.constants` object.  Would be great if it was in the doc.

```
barCodeTypes={[
	Camera.constants.BarCodeType.upce,
	Camera.constants.BarCodeType.ean8,
	Camera.constants.BarCodeType.ean13,
]}
```

grundmanise, 2017-02-02:
Same error:
iOS 10.2
iPhone 5s
`barCodeTypes={['qr']}`

dmarcucci, 2017-02-06:
@grundmanise See @alexmngn's reply. That seems to have fixed the problem for me. Thanks @alexmngn.

+1 for this to be included in the documentation.

grundmanise, 2017-02-07:
Thanks, that works! I overlooked his reply.

dmarcucci, 2017-02-09:
Just FYI - For Android, it expects a `List<String>`. I just set it based on `Platform` to account for this.

Example:
`barCodeTypes={(Platform.OS === 'ios') ? [Camera.constants.BarCodeType.ean13] : ['ean13']}`

You'll need to import `Platform` from `react-native` for this solution.

grundmanise, 2017-02-09:
@dmarcucci Thanks! I also use it like this:

`barCodeTypes={(Platform.OS === 'android') ? ['QR_CODE'] : [Camera.constants.BarCodeType.qr] }`

dmarcucci, 2017-02-10:
@grundmanise Yes. Obviously, you can either do the check for either `ios` or `android` and then set what you need accordingly. I'm only using this for EAN-13, so that's why I have that value there.

alexdunae, 2017-04-04:
Same as:
- https://github.com/lwansbrough/react-native-camera/issues/600
- https://github.com/lwansbrough/react-native-camera/issues/652
- https://github.com/lwansbrough/react-native-camera/issues/499
- https://github.com/lwansbrough/react-native-camera/issues/296

iegik, 2017-09-05:
On `0.10.0` `BarCodeType.qr` will work
https://github.com/lwansbrough/react-native-camera/commit/9162879305aa801bce77a01c4469958fd50d7807

vvakame, 2017-09-29:
I would be happy if this issue write into document. 😉 
It is a problem we can avoid if we know it.

iegik, 2017-12-20:
@vvakame, done

ospfranco, 2018-01-13:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

P.S. There is a new implementation of the camera on android RNCamera, it is on the master branch, you can try it and see if it fixes your problem

closed 2018-01-13 by ospfranco