react-native-camera/react-native-camera #558: Capture result on android not consistent with CameraRoll

danieldkim, 2017-01-18:
### Steps to reproduce
1. Invoke `capture()` method with a `cameraRoll` target. 

### Expected behaviour
The `path` returned should match the `uri` of the objects returned by RN's `CameraRoll.getPhotos()` API.  Should be able to retrieve captured photo object with matching `uri` using `CameraRoll.getPhotos()` immediately after the promise returned by `capture()` is resolved.  This works as expected on iOS.

### Actual behaviour
The `path` in the `capture()` result is the external storage URI, not the URI exposed through the `CameraRoll`, and thus cannot be used to cross reference those objects and retrieve the one captured.  Also, there is an indeterminate delay after the `capture()` promise is resolved before the object appears in the `CameraRoll`.

### Environment
- **React Native version**: 0.40.0
- **React Native platform + platform version**: Android 6.0.1

### react-native-camera
**Version**: `next` branch

I have a fix for this that I will submit a PR for shortly.

npomfret, 2017-03-21:
This should be fixed in https://github.com/lwansbrough/react-native-camera/pull/637, it's added a 2nd field to the response which is the `mediaUri` which hI believe tis the thing you need.

enieber, 2018-08-30:
@sibelius this issue can it is being closed?

n1ru4l, 2018-12-19:
`RNCamera` is deprecated. Please us `RNCamera` instead and open a new issue if you still experience this issue.

closed 2018-12-19 by n1ru4l