react-native-camera/react-native-camera #560: Camera attempts to `setState` after being unmounted

dylanpyle, 2017-01-19:
### Steps to reproduce
1. Render a Camera component 
2. Very quickly stop rendering the component (e.g. to show a loading screen...)

### Expected behaviour
The Camera component recognizes that it's been unmounted and stops trying to do things like `setState`. The [React team outlines some suggestions on how to do this here](https://facebook.github.io/react/blog/2015/12/16/ismounted-antipattern.html).

### Actual behaviour
[image]

### Environment
- **Node.js version**: 6.4.0
- **React Native version**: 0.39.2
- **React Native platform + platform version**: iOS 10

### react-native-camera
**Version**: 0.40.0

dylanpyle, 2017-01-19:
I can't figure out how to get a decent stack trace, but [this is my best guess as to the culprit](https://github.com/lwansbrough/react-native-camera/blob/master/index.js#L160). Since that's `await`ing on the authorization status, it might have mounted - and unmounted again - by the time it finishes. Will follow up with a PR if I get some free time.

ospfranco, 2018-01-13:
Duplicated ticket

Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

P.S. There is a new implementation of the camera on android RNCamera, it is on the master branch, you can try it and see if it fixes your problem

closed 2018-01-13 by ospfranco