react-native-camera/react-native-camera #565: TouchableHighlight inactive after two pictures

chipe13, 2017-01-24:
### Steps to reproduce
1. Take a photo
2. Go back into camera and take another photo
3. One more time and the capture button seems to be disabled

### Expected behaviour
The TouchableHighlight that takes a picture should not be disabled on the third picture. 

### Actual behaviour
After a picture is taken I show the image and have an upload button or retake button. I also have a nav that can take you home. If I either touch retake, or home after taking a pic, and then go back to my camera component, on the third time of taking an image, the button to take the image (TouchableHighlight) the color is a bit faded as if deactive, and does not click. Not sure why it is going inactive after two times of taking a pic

### Environment
- **Node.js version**: 7.2.1
- **React Native version**: 0.40.0
- **React Native platform + platform version**: xCode & simulator -- iOS 10.2

### react-native-camera
camera:^1.0.0-alpha1
npm: 3.10.10

pr-rojan, 2017-02-15:
Hi
I am having same issue. The issue exists only in simulator. Once you run the app in device, it will give you blank screen after third time taking picture.
Did you find any solution with this?
Thank you

chipe13, 2017-02-16:
I did not. Works when I put a dev build on device just fine. Just does not work on simulator. Maybe its a simulator bug that needs an update?

pr-rojan, 2017-02-16:
Can you please share your code?
It might be simulator bug so I tired with updating too but no luck. Actually, everything was working perfectly before, I just started getting this bug.

chipe13, 2017-02-16:
Look here: https://github.com/chipe13/react-native-camera -- components are in the views folder

pr-rojan, 2017-02-16:
Thank you 
I really appreciate it.

pr-rojan, 2017-02-16:
Well i didnt figure out what the issue  was but installing latest version of react-native and its related modules  solved the issue in simulator but still having an issue in device.
Thank you

chipe13, 2017-02-17:
You are having the issue when putting a dev build on your device?

pr-rojan, 2017-02-17:
Yeah, I am able to take picture for first time, when I try to take picture for second time it gives me blank screen.

ospfranco, 2018-01-13:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

P.S. There is a new implementation of the camera on android RNCamera, it is on the master branch, you can try it and see if it fixes your problem

closed 2018-01-13 by ospfranco