react-native-camera/react-native-camera #566: App stops when I try to capture an image.

hudjoubert, 2017-01-24:
### Steps to reproduce
1.Open Camera App
2.Try to capture image
3.

### Expected behaviour
The app should take a picture and console log the result.

### Actual behaviour
The App stop and close when a take a picture
### Environment
- **Node.js version**:latest
- **React Native version**: latest
- **React Native platform + platform version**: Android 5.0, 6.0, 7.0

### react-native-camera
**Version**: npm version

juanluisgv1, 2017-01-25:
Does this happen on Android or iOS or both? I'm facing a similar issue with a "Samsung Galaxy S5 Neo". I'm able to see the camera view but when I take the picture it breaks. On the other hand, it works if I use the front camera and also if I use any other phone (iphone, htc or whatever else).. any idea?

hudjoubert, 2017-01-26:
I managed to make it work on my device (sansung S5), using the front camera, I just added the captureTarget property, but the simulator did not work. As you said now from the other camera I will test here.

hudjoubert, 2017-01-26:
I trying here right now and a I had the same problem. So I dont Know what to do .

juanluisgv1, 2017-01-26:
I must say that it only happens with the Samsung S5. I will research but you don't get any information at all why it breaks...

jlyons, 2017-01-27:
Had this same problem the other day and fixed it by changing quality to 1080p, instead of the default "high".  Seems to be an issue with handing the default, which i assume is why the front camera has no problem.

juanluisgv1, 2017-01-27:
@jlyons you were right! Changing the captureQuality attribute to 1080p or lower it solves the problem 👌🏻

hudjoubert, 2017-01-29:
@jlyons  I was cracking my head trying to know why, thanks for the help

zawarudo, 2017-02-02:
Can confirm this issue on my iPhone 5.

@juanluisgv1 Unfortunately playing with the captureQuality did not fix it for me. Front camera is not even accessible, it shows only black.

**Environment**
Node 7.4.0
React Native 4.0

velozmontero, 2017-02-06:
Im having the same issue with a samsung galaxy s5. The app crashes every time I take a picture with back camera but works fine with front camera. I tried reducing the capture quality but that didn't work for me.

MiLeung, 2017-03-19:
@zawarudo works for me! thanks!

logileifs, 2017-05-11:
I was having the same problem and managed to fix it by adding `android:largeHeap="true"` to my AndroidManifest.xml.

```
<application
      android:name=".MainApplication"
      android:allowBackup="true"
      android:label="@string/app_name"
      android:icon="@mipmap/ic_launcher"
      android:theme="@style/AppTheme"
      android:largeHeap="true">
```

logileifs, 2017-05-11:
Problem seems to be in the `fixOrientation` method, when set to a high capture quality the device runs out of memory and crashes.

npomfret, 2017-05-12:
Have you tried master?  afixOrientation` isn't applied any more (unless you explicitly specify it).  I understand that using `android:largeHeap="true"` isn't recommended (although I can't remember why).

logileifs, 2017-05-12:
I installed the version for RN >0.40. I believe it's version 0.6 and it was still happening there. I'd rather reserve a large heap for my app (although someone does not recommend it) than have it crash

rishimukherjee, 2017-05-17:
I am using OnePlus 3 with Android 7.1.1. None of these work for me. Works fine in Zenfone 2 and iPhone 7 plus though.

heliocaruccio, 2017-09-28:
<key>NSPhotoLibrary**Add**UsageDescription</key>

jifromz, 2017-11-15:
@heliocaruccio - Works for me. Many thanks!

BonnieMilian, 2018-03-09:
I have the same issue, but just on the release apk, on the debugging works perfect, but release is the important one.

```
<Camera
          ref={(cam) => {
            this.camera = cam;
          }}
          style={styles.preview}
          aspect={Camera.constants.Aspect.fill}
          captureQuality={Camera.constants.CaptureQuality["480p"]}
          type={Camera.constants.Type.front}>
          <TouchableOpacity
              style={styles.captureButton}
              onPress={this.takePicture.bind(this)}
          >
            {imgCamera}
          </TouchableOpacity>
        </Camera>
```

```
takePicture() {
    const options = {jpegQuality: 1};
    this.camera.capture({metadata: options})
      .then((data) => {
        //console.log(data);
        this.props.navigation.navigate('DrawRectangle',
          {
            pictureData: data
          }
        );
      })
      .catch(err => console.error(err));
  }
```
The application requires low quality picture.

vygandas, 2018-08-11:
I'm using RNCamera like this
```
takePicture = async () => {
        const options: TakePictureOptions = {
            // quality: 0.9,
            base64: true,
            exif: false,
            // width?: number;
            // mirrorImage?: boolean;

            /** Android only */
            // skipProcessing?: boolean;
            /** Android only */
            fixOrientation: true,

            /** iOS only */
            forceUpOrientation: false
        };
        this.camera.takePictureAsync(options)
            .then(data => console.log(data.base64))
            .catch(error => console.log(error));
    };
render() {
        return (
            <View style={styles.container}>
                <RNCamera
                    ref={cam => {
                        this.camera = (cam as RNCamera);
                    }}
                    style={styles.preview}
                    type={RNCamera.Constants.Type.back}
                >
                    <TouchableHighlight onPress={this.takePicture}>
                        <Image
                            style={{ flex: 0, width: 100, height: 100 }}
                            source={require('./images/cameraButton.png')}
                        />
                    </TouchableHighlight>
                </RNCamera>
            </View>
        );
    }
```

when `type={RNCamera.Constants.Type.front}` it works. If I use `back` it fails. App closes and says "Unfortunately app has stopped....."

I've added things to manifest like exif 27+, android:largeHeap="true" and other things but it doesn't help.

n1ru4l, 2018-12-19:
Please open a new issue with crashlogs if you still experience this issue with the newest version and `RNCamera`.

closed 2018-12-19 by n1ru4l