react-native-camera/react-native-camera #568: Is there any way to set the zoom value?

peterchermie, 2017-01-26:
Hi, 

I need to create a zoom slider, is there any method to set the zoom value or bring the native zoom slider to the camera viewfinder?

Many thanks in advance,
Pete.r

Marthyn, 2017-02-09:
I would also like to do this.

peterchermie, 2017-02-09:
hi @Marthyn, I have added that functionality, please have a look here: https://github.com/peterchermie/react-native-camera/commit/d55113ff3d37e25665482f6b9511297e962c27b5

Marthyn, 2017-02-09:
@peterchermie That's awesome! So then you just call the method setZoom on the camera? Like:

```
zoomBySlider = (zoomLevel:number) => {
    if (this.camera) {
      this.camera.setZoom(zoomLevel);
    }
```

And do you also know what zoomlevels there are available?

peterchermie, 2017-02-09:
@Marthyn yup, you just call `this.camera.setZoom(zoomLevel)`

`zoomLevel` can be anything from 1 to 16.

I use it with the slider, like this: 
`<Slider
						value={0}
						minimumValue={1}
						maximumValue={16}
						onValueChange={value => this.camera.setZoom(value)}
					/>`

I hope that helps 😄

Marthyn, 2017-02-09:
You made my day @peterchermie 👍

sibelius, 2018-01-23:
fixed on master

closed 2018-01-23 by sibelius