react-native-camera/react-native-camera #569: [Android] Battery drain issue

kevinnguy, 2017-01-27:
Has anyone come across having a battery issue on their Android devices? When I'm using the camera for an extended period of time (10-30 minutes), my battery drains unexpectedly fast. I'm using the camera to scan barcodes and was wondering if there are any ways of improving camera/battery performance. Thanks!

### Environment
- **Node.js version**: 3.10.10
- **React Native version**: 0.40
- **React Native platform + platform version**: Android 5.0

### react-native-camera
**Version**: 0.5.1

cfournel, 2017-05-22:
As mentionned in https://github.com/lwansbrough/react-native-camera/issues/460 , the only way is to turn off camera by not displaying the view.

enieber, 2018-08-30:
@sibelius  this issue can it is being closed?

closed 2018-09-04 by kevinnguy