react-native-camera/react-native-camera #570: creating a test with jest

agrcrobles, 2017-01-27:
### Steps to reproduce

I am testing my react-native-app with `jest` + `react-test-renderer` by doing the following:
```
import 'react-native';
import React from 'react';
import Index from '../index.ios.js';

import renderer from 'react-test-renderer';

it('renders correctly', () => {
  const tree = renderer.create(
    <Index />
  );
});
```
And getting the following exception:

> TypeError: Cannot read property 'Aspect' of undefined

### Expected behaviour
Should test pass

### Actual behaviour
I think there is an issue in the [constructor](https://github.com/agrcrobles/react-native-camera/blob/master/index.js#L18) that throws an exception when try requiring native component in a test environment rather than lazy loading it.

### Environment
- **Node.js version**: 6.9.4
- **React Native version**: 0.40.0
### react-native-camera
**Version**: latest

sibelius, 2017-01-27:
u need to mock this package

agrcrobles, 2017-01-27:
To fix that I created `react-native-camera` react component in `__mocks__` folder:

```
import React from 'react';

const constants = constants = {
  Aspect: {},
  BarCodeType: {},
  Type: {},
  CaptureMode: {},
  CaptureTarget: {},
  CaptureQuality: {},
  Orientation: {},
  FlashMode: {},
  TorchMode: {}
};

class Camera extends React.Component {

  static constants = constants
  render() {
    return null;
  }
}

Camera.constants = constants;

export default Camera;
```

And mocked it like this in the test case:

`jest.mock('react-native-camera', () => mockCamera);`

But this could be a workaround depending on what and how deep I need to go with the test.

skovhus, 2017-04-20:
I think we should close this issue.

UgurcanOzdemir, 2017-05-19:
@agrcrobles your mock working but now I have another error
```TypeError: Cannot read property 'default' of undefined```

assafb81, 2017-10-10:
I encounter few issues with react-native-camera mock.
the main issue is that I am getting the following error:
```
? Test suite failed to run

TypeError: Cannot read property 'default' of undefined

     at Object.user;c:\**\node_modules\react-native-camera\index.js; (__tests__/index.android.js:10:70)
     at Object.<anonymous> (src/components/fileA.js:5:24)
     at Object.<anonymous> (src/containers/fileB.js:11:20)
```

the second issue is multiple ESLINT errors when writing: 
```
const constants = constants = {
    Aspect: {},
    BarCodeType: {},
    Type: {},
    CaptureMode: {},
    CaptureTarget: {},
    CaptureQuality: {},
    Orientation: {},
    FlashMode: {},
    TorchMode: {},
};
```

'constants' is constants (no-const-assign)
unexpected chained assignment. (no-multi-assign)
'constants' was used before it was defined. (no-use-before-define)

this seems to solve the problem: `const constants = {...}`
instead of `const constants = constants = {...}`

I am using, currently, 3 properties:
`style, aspect and onBarCodeRead()`. I don't see any reference to `style` and `onBarCodeRead` property in the example mock component.

I am using:
```
"react": "16.0.0-alpha.12",
"react-native": "0.45.1",
"react-native-camera": "^0.10.0",
"jest": "20.0.4",
```

assafb81, 2017-10-10:
@sibelius - why close an open question?

agrcrobles, 2017-10-10:
@assafb81 Find the question [here](https://stackoverflow.com/questions/41811477/how-to-mock-react-native-camera-with-jest)

assafb81, 2017-10-10:
When adding the code above - as @agrcrobles suggested on Jan 27 - I've encountered another problem which I asked today.

trumpetx, 2017-10-16:
I'm having the exact same problems that @assafb81 is having.  I think it's important to get a good testing recipe for this.  I'm not sure it matters the strategy (mocking, etc) - it just needs to work and be part of the docs IMHO.

assafb81, 2017-10-19:
I tried this:
`jest.mock('react-native-camera', () => require.requireActual('../__mocks__/react-native-camera').default()` in my test:
```
import 'react-native';
import React from 'react';
import Index from '../app/index'; //my awesome application that contains the react native camera

import renderer from 'react-test-renderer';
jest.mock('react-native-camera', () => require.requireActual('../__mocks__/react-native-camera').default();

it('renders correctly', () => {
  const tree = renderer.create(
    <Index />,
  );
});
```
And I get this error:
```
FAIL  __tests__\index.ios.js
 ? Test suite failed to run

   TypeError: Cannot call a class as a function

     at _classCallCheck (__mocks__/react-native-camera.js:1:903)
     at Object.Camera [as default] (__mocks__/react-native-camera.js:15:88)
     at Object.user;C:\Projects\**\App\__mocks__\react-native-camera.js; (__tests__/index.ios.js:15:116)
     at Object.<anonymous> (src/components/componentA.js:5:24)
     at Object.<anonymous> (src/containers/ContainerA.js:11:20)
```

atillaordog, 2017-10-27:
Go the solution for the error:

`TypeError: Cannot read property 'default' of undefined`

Simply change this line
`import mockCamera from '../__mock__/react-native-camera';`
to
`import * as mockCamera from '../__mock__/react-native-camera';`

This worked for me so I could use the mock to successfully test my components using camera component

assafb81, 2017-10-31:
@atillaordog, I tried your solution and I changed:

`jest.mock('react-native-camera', () => require.requireActual('../__mocks__/react-native-camera').default());`

to:
```
import * as mockCamera from '../__mocks__/react-native-camera';
jest.mock('react-native-camera', () => mockCamera);
```

and I got the following error:
```
 ? Test suite failed to run

   TypeError: Cannot call a class as a function

     at _classCallCheck (__mocks__/react-native-camera.js:1:903)
     at Object.Camera [as default] (__mocks__/react-native-camera.js:15:88)
     at Object.user;c:\**\App\__ mocks __\react-native-camera.js; (__tests__/index.ios.js:10:70)
     at Object.<anonymous> (src/components/componentA.js:5:24)
     at Object.<anonymous> (src/containers/componentB.js:11:20)

```

atillaordog, 2017-10-31:
My Camera.js file in the mocks looks like: 

```js
import React from 'react';

const constants = constants = {
  Aspect: {},
  BarCodeType: {},
  Type: {},
  CaptureMode: {},
  CaptureTarget: {},
  CaptureQuality: {},
  Orientation: {},
  FlashMode: {},
  TorchMode: {}
};

class Camera extends React.Component {

  static constants = constants
  render() {
    return null;
  }
}

Camera.constants = constants;

export default Camera;
```
And I have the following versions in my package.json
 "enzyme": "^2.9.1",
  "jest": "20.0.4"

assafb81, 2017-10-31:
my code is almost identical.
I get an ESLint error when I write this line: 
`const constants = constants = {`
so I changed it to:
`const constants = {`

my current versions are:
```
"enzyme": "^3.1.0",
"jest": "21.2.1",
```

But I upgraded them only yesterday and I still got the same error with the older versions

atillaordog, 2017-10-31:
Hmm, I do not know what could be the problem, maybe your jest settings in package.json?
"jest": {
    "preset": "react-native"
}

I did not try to reproduce your error, since it works on my side, but if I meet with this problem and I get a solution, I will post it.

assafb81, 2017-11-20:
did you add to your package.json file, under jest section the following:
```
"jest": {
    ...
    "setupFiles": [
        "<rootDir>/__mocks__/react-native-camera",
        ...
    ]
}
```

assafb81, 2017-11-20:
@agrcrobles Or @sibelius  - can you re-open the issue?

I also tried to add to package.json file under Jest section a reference to the mock component:

```
"jest": {
    "setupFiles": [
        "<rootDir>/__mocks__/react-native-camera",
        ...
    ]
}
```

JohannesWi, 2019-09-09:
any solution for mocking "onBarcodeRead"?