react-native-camera/react-native-camera #571: [Feature] Add real-time options to camera

kresic-ivan-nsoft, 2017-01-27:
Hi guys,

would it be possible to add real-time options for camera controls, triggering the flash, switch to front camera or similar?

tobiaskiel, 2017-02-06:
It already is
` <Camera ref={(cam) => { this.camera = cam; } }
          style={styles.preview}
          type={this.state.cameraType}
          captureQuality={this.state.quality}
          torchMode={this.state.torchOn}
          aspect={Camera.constants.Aspect.fill}
          captureTarget={Camera.constants.CaptureTarget.cameraRoll }
        >`
Then just update required state for toggeling options like, type and torchMode

kresic-ivan-nsoft, 2017-02-08:
Great, thanks for the tip! Any way to apply filters to the camera view (black & white for example)?

hedgerh, 2017-07-18:
@kresic-ivan-nsoft: ooh, itd be awesome to be able to apply filters.  you can do it with images pretty easily, but filtering the live preview from the `<Camera>` component could be tricky.  for info about filters, check something like this out https://www.npmjs.com/package/image-filter-core

closed 2018-09-14 by SimonErm