react-native-camera/react-native-camera #572: Use of undeclared identifier RCTErrorUnspecified

grahamtallen, 2017-01-27:
### Steps to reproduce
1. Build and Run in Xcode

### Expected behaviour
The app builds for the ios simulator

### Actual behaviour
Error logs 8 times: 
Error in RCTCameraManger.m:
"Use of undeclared identifier 'RCTErrorUnspecified'"

### Environment
- node 6.2.2
- React-Native 0.38.0
- iOS 9.0,

### react-native-camera
**Version**: 0.4.1

umer990, 2017-01-30:
Hi Graham! Any luck? I am get same error!
/Umer

bharath-kamath, 2017-05-04:
@umer990 @grahamtallen  Any luck with this, guys?

frmrgr, 2017-05-10:
This is because you tried to use this library via cocoapods, but did not move react-native itself into Podfile, so react-native-camera pod installed default React pod as a dependency, which is of version 0.11, which is too old. Your options:
1) Do not use it as cocoapod, just react-native-link it into the project.
2) Find manual how to use react-native via pods (you have to do pod 'React' with a lot of subspecs and remove those libraries from your main project etc.). Add pod 'react-native-camera' as described here and DO NOT react-native link it, otherwise you'll have duplicate symbols in linker.

I'd stick to option (1), because pods suck for react-native.

bharath-kamath, 2017-05-11:
I had not used it as cocoapod, had directly installed it via npm. I think there was a version mismatch that was causing the problem.

ashusdn, 2017-08-28:
Hello Team,

I am facing the same issue. Actually, I am getting an error on below line when trying to build:

   reject(RCTErrorUnspecified, nil, RCTErrorWithMessage(error.description));

My details:  "react-native": "0.45.1",
 "react-native-camera": "^0.10.0",

I have follow same instruction as mentioned on https://github.com/lwansbrough/react-native-camera.

Can you please provide any solution regarding this.

maurizio96, 2018-01-10:
Hi all,
I am facing th esame issue. Is it possible that the cause is a conflict with the _react-native-fcm_ module?

To make the react-native-fcm module work correctly, I'm forced to use the .xcworkspace file.
Can this cause this problem?

```
"react-native": "0.51.0",
"react-native-fcm": "^11.0.2",
"react-native-camera": "^0.12.0",
```

Am I doing something wrong?

Thank you

sibelius, 2018-01-10:
it looks like this is a problem with pod, open another issue to track for pod installations

closed 2018-01-10 by sibelius