react-native-camera/react-native-camera #581: Camera randomly has incorrect orientation

RuairiOliver, 2017-02-02:
**Steps to reproduce**
Display camera in landscape mode. All displays fine.
Capture a photo and change state on the view

**Expected behaviour**
I can take another photo with the same orientation

**Actual behaviour**
At random times the image displays in portrait while I'm taking the photo. Eventually it corrects itself. Setting a fixed orientation doesn't help and using the front or back camera on the iPhone has the same problem.

Using latest version of react-native-camera(0.6) and on version 0.4 of React Native.
This was happening on the previous version of this library and react native though.

Marthyn, 2017-02-09:
I have had reports of the same.

kaifaust, 2017-03-21:
We're experiencing the same problem: iPad Air 2 in landscape mode. It displays the wrong orientation 30% of the time and it fixes itself in 10-12 seconds after initialization every time. Very strange behavior. We're having to implement a hack to mount — and keep mounted — the camera only once in the background during the use of the app. This is obviously problematic, but until there's an Objective-C angel out there who commits a fix, this will have to do.

hugohow, 2017-03-29:
Ok I get the exact same problem, I will try your solution ;)

RuairiOliver, 2017-03-30:
To solve it in the end I changed to a new view to display the picture I took. Then changed to the view that held the camera component again if I needed to take another photo. Works fine without changing state. 

Also I'm keeping it fixed on the front camera, turned off the option to flip.

RazerTang, 2017-07-25:
Have you  solved this problem? @hugohow

hugohow, 2017-07-25:
Yep @RazerTang the method of @kaifaust works. It's a lot of changes in the code, but it works.

instancewoo, 2017-10-13:
in RCTCameraManager.m
- (id)init{
//.... do the following modification
  self.sessionQueue = dispatch_get_main_queue();
            //dispatch_queue_create("cameraManagerQueue", DISPATCH_QUEUE_SERIAL);
}

craigcoles, 2017-10-16:
I have been testing the fix that @instancewoo suggested and it seems to stop issues occurring. Would this be a viable solution going forward?

lakhman, 2017-12-16:
@instancewoo's fix works for me also, i'm on latest master of RNC. +1 Merge to help others.

mik4el, 2018-01-18:
@instancewoo thanks, solves my problem as well

ffiargus, 2018-01-18:
@instancewoo's fix solves the problem, but now the screen with camera component is slower to navigate too, any solutions?

Rewieer, 2018-01-22:
The same problem also occurs in Android.

Nauzer, 2018-03-01:
Anyone has found a fix for this issue on RNCamera?

Progoogler, 2018-03-19:
Have you tried `takePictureAsync({forceUpOrientation: true, fixOrientation: true})`?

eliaahadi, 2018-03-24:
@Progoogler I tried that and it still rotates incorrectly. Any other suggestions?

Friendly-Robot, 2018-03-24:
@eliaahadi Strange, it worked for me out of the box. Maybe it's the specific device you're testing with that's incompatible. If those don't work maybe consider using react-native-image-resizer to allow the user to manually rotate the image. Otherwise you can look into the `rotate(exif Orientation)` function in MutableImage.java. That's where the rotation happens for Android at least.

mgela, 2018-05-09:
@Friendly-Robot thanks, worked for me.

SimonErm, 2018-09-17:
Can you try if it is fixed with master?

n1ru4l, 2018-12-12:
Closed during to missing feedback.

closed 2018-12-12 by n1ru4l