react-native-camera/react-native-camera #584: undefined is not an object (evaluating 'CameraManager.Aspect')

subhash8108, 2017-02-03:
I  have simple react native run from my emulator, and your plugin (component) install successfully, but one function not push on my code. 

please find attached file.
[image]

Please help me... how to resolved this issue.
I'm waiting your reply.

Thanks
Subhash

gustarus, 2017-02-03:
Same problem I've got today 😭.

gustarus, 2017-02-03:
Resolved: `react-native link react-native-camera` one more time.

ptomasroos, 2017-02-09:
Should we close, is the workaround working?

AlisonWonderlandApps, 2017-02-09:
I have the same error

ptomasroos, 2017-02-13:
Then I will close. Thanks!

closed 2017-02-13 by ptomasroos

trops, 2017-02-17:
I linked and info.plist doesn't have Camera permissions. Just installed first time today. 
"Cannot read property "Aspect" of undefined from example code.

yinoneliraz, 2017-03-18:
Can you please write steps to solve this in a more elaborated way?

muhlucas, 2017-03-27:
I had same issue
My solution was react-native link react-native-camera put Camera permission on info.plist 
And recompile react native run-ios

matthieucan, 2017-04-18:
On Android, uninstalling and reinstalling the apk worked for me.

rcorrie, 2017-06-28:
The key for me on iOS was

```
rm -rf ./node_modules
npm install
watchman watch-del-all
```

then I shut down packager, quit simulator, then `react-native run-ios`

mashwinprdxn, 2018-06-21:
In case anyone's facing this issue on Android, please try the [gradle upgrade guide](https://github.com/react-native-community/react-native-camera/blob/master/docs/GradleUpgradeGuide.md). It worked for me.